project-chip/connectedhomeip #24440: [BUG] [nrfconnect] google home app cannot finish commissioning

mmt-10, 2023-01-16:
### Reproduction steps

1. set google nest hub gen2(preview program) and google home app(public preview) on android
2. flash `lighting app` for nrfconnect
  ```
  $ west build -b ubx_evkninab3_nrf52840 -- -DCONF_FILE=prj_no_dfu.conf
  $ west flash --erase -r pyocd
```
3. run `google home app` and start commissioning
`google home app` shows error message.

Apple Home finished commissioning normally and can control this lighting app cluster.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit b4c2827465b0b36caa231fa8d055ce128194db00 (HEAD -> master, origin/master, origin/HEAD)

### Platform

nrf

### Anything else?

### Environment
ubuntu 20.04
target board : ublox EVK NINA-B302

nrf version : commit 188a16036ab968bc0f9226ba178ca461548a8c66 (HEAD, tag: v2.2.0)
zephyr version : commit c0689b16ff127d1b71a4cc20310d476e1807e26e (HEAD, tag: v3.2.99-ncs1-rc2, tag: v3.2.99-ncs1, manifest-rev)
gcc arm version : gcc-arm-none-eabi-10.3-2021.10

FHFS, 2023-01-25:
I have had a similar problem. I was using the default ProductID(0xfff1) and VendorID(0x8005) for my nRF52840 device. I had a different vendorid(0xfff4) and productid(0xBEEF) on [DeveloperConsole](https://console.home.google.com/projects/my-project/matter)

I was unable to commission my device after removing it.

After I added the correct productid and vendorid to the developerconsole I was able to add my devices again.

s07641069, 2023-02-03:
Same for Telink. Lighting app works well, but for example Light-Switch app starts to fail with Google Nest Hub.
Used for test old bins which works fine previously. Looks like some changes on Google Home or Google Hest Hub firmware.
Same bins can be successfully paired with Apple Home and Apple HomPod.

mmt-10, 2023-02-04:
Hi,
I think it may be google home issue., too
So, I posted it to below google forum.
[https://www.googlenestcommunity.com/](https://www.googlenestcommunity.com/)

mmt-10, 2023-02-08:
I checked log again.
`Moving from state 'kEstablishing' --> 'kActive'`, then type 0001 occurs.
As my understanding, `type 0001` is error.

```
[00:00:29.747,985] <inf> chip: [IN](U) Sending msg 31115259 to IP address 'BLE'
[00:00:29.749,328] <inf> chip: [SC]SecureSession[0x20003068]: Moving from state 'kEstablishing' --> 'kActive'
[00:00:29.749,481] <dbg> chip: LogV: [IN]SecureSession[0x20003068]: Activated - Type:1 LSID:10
[00:00:29.749,694] <dbg> chip: LogV: [IN]New secure session activated for device <FFFFFFFB00000000, 0>, LSID:10 PSID:37730!
[00:00:29.749,816] <inf> chip: [SVR]Commissioning completed session establishment step
[00:00:29.750,335] <inf> chip: [DIS]Updating services using commissioning mode 0
[00:00:29.751,281] <inf> chip: [DIS]Advertise commission parameter vendorID=65521 productID=32773 discriminator=3840/15 cm=0
[00:00:29.751,617] <err> chip: [DIS]Failed to advertise extended commissionable node: 3
[00:00:29.751,739] <dbg> chip: LogV: [DIS]Scheduling extended discovery timeout in 900s
[00:00:29.751,861] <err> chip: [DIS]Failed to finalize service update: 1c
[00:00:29.752,014] <inf> chip: [SVR]Device completed Rendezvous process
[00:00:29.837,615] <inf> chip: [EM]>>> [E:19389r M:82804893] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
[00:00:29.837,768] <dbg> chip: LogV: [IM]Received Read request
[00:00:29.839,782] <inf> chip: [EM]<<< [E:19389r M:55182754] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
[00:00:29.839,935] <inf> chip: [IN](S) Sending msg 55182754 on secure session with LSID: 10
[00:00:29.927,581] <inf> chip: [EM]>>> [E:19390r M:82804894] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
[00:00:29.927,703] <dbg> chip: LogV: [IM]Received Read request
[00:00:29.929,748] <inf> chip: [EM]<<< [E:19390r M:55182755] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
[00:00:29.929,870] <inf> chip: [IN](S) Sending msg 55182755 on secure session with LSID: 10
[00:00:30.017,639] <inf> chip: [EM]>>> [E:19391r M:82804895] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
[00:00:30.017,761] <dbg> chip: LogV: [IM]Received Read request
[00:00:30.019,805] <inf> chip: [EM]<<< [E:19391r M:55182756] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
[00:00:30.019,958] <inf> chip: [IN](S) Sending msg 55182756 on secure session with LSID: 10
[00:00:30.107,482] <inf> chip: [EM]>>> [E:19392r M:82804896] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
[00:00:30.107,574] <dbg> chip: LogV: [IM]Received Read request
[00:00:30.109,619] <inf> chip: [EM]<<< [E:19392r M:55182757] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
[00:00:30.109,771] <inf> chip: [IN](S) Sending msg 55182757 on secure session with LSID: 10
[00:00:30.287,078] <inf> chip: [EM]>>> [E:19393r M:82804897] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
[00:00:30.287,200] <dbg> chip: LogV: [IM]Received Read request
[00:00:30.289,245] <inf> chip: [EM]<<< [E:19393r M:55182758] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
[00:00:30.289,398] <inf> chip: [IN](S) Sending msg 55182758 on secure session with LSID: 10
[00:00:30.377,593] <inf> chip: [EM]>>> [E:19394r M:82804898] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
[00:00:30.377,716] <dbg> chip: LogV: [IM]Received Read request
[00:00:30.383,270] <inf> chip: [EM]<<< [E:19394r M:55182759] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
[00:00:30.383,422] <inf> chip: [IN](S) Sending msg 55182759 on secure session with LSID: 10
[00:00:30.917,480] <inf> chip: [EM]>>> [E:19395r M:82804899] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
[00:00:30.917,602] <dbg> chip: LogV: [IM]Received Read request
[00:00:30.919,616] <inf> chip: [EM]<<< [E:19395r M:55182760] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
[00:00:30.919,799] <inf> chip: [IN](S) Sending msg 55182760 on secure session with LSID: 10
[00:00:32.357,025] <inf> chip: [EM]>>> [E:19396r M:82804900] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:0a (IM:TimedRequest)
[00:00:32.358,001] <inf> chip: [EM]<<< [E:19396r M:55182761] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:01 (IM:StatusResponse)
[00:00:32.358,154] <inf> chip: [IN](S) Sending msg 55182761 on secure session with LSID: 10
[00:00:32.447,601] <inf> chip: [EM]>>> [E:19396r M:82804901] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
[00:00:32.448,272] <inf> chip: [FS]GeneralCommissioning: Received ArmFailSafe (120s)
[00:00:32.449,615] <inf> chip: [EM]<<< [E:19396r M:55182762] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
[00:00:32.449,768] <inf> chip: [IN](S) Sending msg 55182762 on secure session with LSID: 10
[00:00:32.537,567] <inf> chip: [EM]>>> [E:19397r M:82804902] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
[00:00:32.537,658] <dbg> chip: LogV: [IM]Received Read request
[00:00:32.539,703] <inf> chip: [EM]<<< [E:19397r M:55182763] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
[00:00:32.539,855] <inf> chip: [IN](S) Sending msg 55182763 on secure session with LSID: 10
[... cut ...]
[00:00:34.293,029] <inf> chip: [EM]>>> [E:19402r M:82804908] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
[00:00:34.293,670] <inf> chip: [ZCL]OpCreds: Received a CSRRequest command
[00:00:34.497,344] <inf> chip: [ZCL]OpCreds: AllocatePendingOperationalKey succeeded
[00:00:34.528,686] <inf> chip: [ZCL]OpCreds: CSRRequest successful.
[00:00:34.531,341] <inf> chip: [EM]<<< [E:19402r M:55182769] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
[00:00:34.531,494] <inf> chip: [IN](S) Sending msg 55182769 on secure session with LSID: 10
```

closed 2023-10-15 by stale[bot]