project-chip/connectedhomeip #24445: [BUG] Does Amazon Alex support Bridged devices? Pairing with dynamic-bridge-app failed

Apollon77, 2023-01-16:
### Reproduction steps

Does anyone know if Amazon Alexa should support bridged devices with Matter or only simple devices?

If I try to pair Amazon Alexa with "dynamic-bridge-app" then it fails (roughly the same oif I try with my development version of node-matter :-) )

```
[1673868783251] [20318:17564122] [DL] KVS will be written to: file:///tmp/chip_kvs
[1673868783267] [20318:17564122] [DL] _Init
[1673868783267] [20318:17564122] [SPT] *** WARNING: Using temporary passcode 20202021 due to no neither --passcode or --spake2p-verifier-base64 given on command line. This is temporary and will disappear. Please update your scripts to explicitly configure onboarding credentials. ***
[1673868783267] [20318:17564122] [SPT] *** WARNING: Using temporary test discriminator 3840 due to --discriminator not given on command line. This is temporary and will disappear. Please update your scripts to explicitly configure discriminator. ***
[1673868783267] [20318:17564122] [SPT] PASE PBKDF iterations set to 1000
[1673868783267] [20318:17564122] [SPT] LinuxCommissionableDataProvider didn't get a PASE salt, generating one.
[1673868783271] [20318:17564122] [DL] Device Configuration:
[1673868783271] [20318:17564122] [DL]   Serial Number: TEST_SN
[1673868783271] [20318:17564122] [DL]   Vendor Id: 65521 (0xFFF1)
[1673868783271] [20318:17564122] [DL]   Product Id: 32769 (0x8001)
[1673868783271] [20318:17564122] [DL]   Hardware Version: 0
[1673868783271] [20318:17564122] [DL]   Setup Pin Code (0 for UNKNOWN/ERROR): 20202021
[1673868783271] [20318:17564122] [DL]   Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 3840 (0xF00)
[1673868783271] [20318:17564122] [DL]   Manufacturing Date: (not set)
[1673868783271] [20318:17564122] [DL]   Device Type: 65535 (0xFFFF)
[1673868783271] [20318:17564122] [-] ==== Onboarding payload for Standard Commissioning Flow ====
[1673868783271] [20318:17564122] [SVR] SetupQRCode: [MT:-24J042C00KA0648G00]
[1673868783271] [20318:17564122] [SVR] Copy/paste the below URL in a browser to see the QR Code:
[1673868783271] [20318:17564122] [SVR] https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A-24J042C00KA0648G00
[1673868783271] [20318:17564122] [SVR] Manual pairing code: [34970112332]
[1673868783271] [20318:17564122] [DL] _SetDeviceName
[1673868783271] [20318:17564122] [DL] _SetAdvertisingEnabled
[1673868783271] [20318:17564122] [SVR] Server initializing...
[1673868783271] [20318:17564122] [FP] Initializing FabricTable from persistent storage
[1673868783271] [20318:17564122] [TS] Last Known Good Time: 2023-01-16T12:17:49
[1673868783272] [20318:17564122] [DMG] AccessControl: initializing
[1673868783272] [20318:17564122] [DMG] Examples::AccessControlDelegate::Init
[1673868783272] [20318:17564122] [DMG] AccessControl: setting
[1673868783272] [20318:17564122] [DMG] DefaultAclStorage: initializing
[1673868783272] [20318:17564122] [DMG] DefaultAclStorage: 0 entries loaded
[1673868783272] [20318:17564122] [IN] UDP::Init bind&listen port=5540
[1673868783272] [20318:17564122] [IN] UDP::Init bound to port=5540
[1673868783272] [20318:17564122] [IN] UDP::Init bind&listen port=5540
[1673868783272] [20318:17564122] [IN] UDP::Init bound to port=5540
[1673868783272] [20318:17564122] [IN] BLEBase::Init - setting/overriding transport
[1673868783272] [20318:17564122] [IN] TransportMgr initialized
[1673868783274] [20318:17564122] [ZCL] Using ZAP configuration...
[1673868783274] [20318:17564122] [DMG] Failed to read stored attribute (0, 0x0000_0028, 0x0000_0005: ../../third_party/connectedhomeip/src/platform/Darwin/KeyValueStoreManagerImpl.mm:222: CHIP Error 0x000000A0: Value not found in the persisted storage
[1673868783274] [20318:17564122] [DMG] Failed to read stored attribute (0, 0x0000_0028, 0x0000_0010: ../../third_party/connectedhomeip/src/platform/Darwin/KeyValueStoreManagerImpl.mm:222: CHIP Error 0x000000A0: Value not found in the persisted storage
[1673868783274] [20318:17564122] [DMG] Failed to read stored attribute (0, 0x0000_002C, 0x0000_0000: ../../third_party/connectedhomeip/src/platform/Darwin/KeyValueStoreManagerImpl.mm:222: CHIP Error 0x000000A0: Value not found in the persisted storage
[1673868783274] [20318:17564122] [DMG] Failed to read stored attribute (0, 0x0000_002C, 0x0000_0001: ../../third_party/connectedhomeip/src/platform/Darwin/KeyValueStoreManagerImpl.mm:222: CHIP Error 0x000000A0: Value not found in the persisted storage
[1673868783274] [20318:17564122] [DMG] Failed to read stored attribute (0, 0x0000_002D, 0x0000_0000: ../../third_party/connectedhomeip/src/platform/Darwin/KeyValueStoreManagerImpl.mm:222: CHIP Error 0x000000A0: Value not found in the persisted storage
[1673868783275] [20318:17564122] [DMG] Failed to read stored attribute (2, 0x0000_0008, 0x0000_4000: ../../third_party/connectedhomeip/src/platform/Darwin/KeyValueStoreManagerImpl.mm:222: CHIP Error 0x000000A0: Value not found in the persisted storage
[1673868783275] [20318:17564122] [DMG] AccessControlCluster: initializing
[1673868783275] [20318:17564122] [ZCL] Initiating Admin Commissioning cluster.
[1673868783275] [20318:17564122] [DMG] Endpoint 0, Cluster 0x0000_0030 update version to 745f2fb2
[1673868783275] [20318:17564122] [ZCL] GeneralDiagnostics: OnDeviceReboot
[1673868783275] [20318:17564122] [DMG] Endpoint 0, Cluster 0x0000_0033 update version to 688fd1f8
[1673868783275] [20318:17564122] [EVL] LogEvent event number: 0x0000000000020000 priority: 2, endpoint id:  0x0 cluster id: 0x0000_0033 event id: 0x3 Sys timestamp: 0x0000000086617D44
[1673868783275] [20318:17564122] [DMG] Endpoint 0, Cluster 0x0000_002B update version to f55be764
[1673868783276] [20318:17564122] [DMG] Endpoint 2, Cluster 0x0000_0008 update version to 88633445
[1673868783276] [20318:17564122] [IN] SecureSession[0x600001e6c000]: Allocated Type:1 LSID:56125
[1673868783276] [20318:17564122] [SC] Assigned local session key ID 56125
[1673868783276] [20318:17564122] [SC] Waiting for PBKDF param request
[1673868783276] [20318:17564122] [DL] _SetAdvertisingEnabled
[1673868783276] [20318:17564122] [SVR] BLE networking available but BLE advertising is not supported
[1673868783276] [20318:17564122] [DIS] Updating services using commissioning mode 1
[1673868783276] [20318:17564122] [DL] Using wifi MAC for hostname
[1673868783276] [20318:17564122] [DIS] Advertise commission parameter vendorID=65521 productID=32769 discriminator=3840/15 cm=1
[1673868783276] [20318:17564122] [DIS] Registering service D0FA56F466C1171C on host 000000000000.local. with port 5540 and type: _matterc._udp,_V65521,_S15,_L3840,_CM on interface id: 0
[1673868783276] [20318:17564122] [DIS] Updating services using commissioning mode 1
[1673868783277] [20318:17564122] [DL] Using wifi MAC for hostname
[1673868783277] [20318:17564122] [DIS] Advertise commission parameter vendorID=65521 productID=32769 discriminator=3840/15 cm=1
[... 582 lines cut ...]
[1673868806393] [20318:17564444] [DIS] Mdns: OnRegisterRecord flags: 1
[1673868806393] [20318:17564444] [DIS] Mdns: OnRegisterRecord flags: 1
[1673868806393] [20318:17564444] [DIS] Mdns: OnRegisterRecord flags: 1
[1673868806393] [20318:17564444] [DIS] Mdns: OnRegisterRecord flags: 1
[1673868806393] [20318:17564444] [DIS] Mdns: OnRegisterRecord flags: 1
[1673868806393] [20318:17564444] [DIS] Mdns: OnRegisterRecord flags: 1
[1673868806393] [20318:17564444] [DIS] Mdns: OnRegisterRecord flags: 1
[1673868806393] [20318:17564444] [DIS] Mdns: OnRegisterRecord flags: 1
[1673868806393] [20318:17564444] [DIS] Mdns: OnRegisterRecord flags: 1
[1673868806393] [20318:17564444] [DIS] Mdns: OnRegisterRecord flags: 1
[1673868806394] [20318:17564444] [DIS] Mdns: OnRegisterRecord flags: 0

```

### Bug prevalence

Bridged Devices / Amazon Alexa

### GitHub hash of the SDK that was being used

b4c2827

### Platform

other

raven-worx, 2023-01-17:
AFAIK Amazon Alexa only supports lights and outlets with Matter so far.
Support for more device types will follow with updates.

Apollon77, 2023-01-17:
I assumed that ... I think it would make a huge sense to have a public" overveiw on what exactly is supported by "the big ones"? :-(

closed 2023-01-17 by Apollon77