project-chip/connectedhomeip #24447: [BUG] Does iOS 16.2 already support bridged devices? (dynamic-bridge-app)

Apollon77, 2023-01-16:
### Reproduction steps

Maybe not really a "bug" (unless it should work already)

I tried around with "dynamic-bridge-app" and IOS 16.2 device. Pairing was possible but afterwards I got ONE Matter device in Apple Home which was shown as "not supported"

Here the pairing log:
[bridge-pair-ios.txt](https://github.com/project-chip/connectedhomeip/files/10425742/bridge-pair-ios.txt)

### Bug prevalence

iOS 16.2, Bridged Devices

### GitHub hash of the SDK that was being used

b4c2827

### Platform

core

bndkk, 2023-01-16:
In my experience the Home App says "not supported" when there is no valid endpoint added to the bridge. An endpoint is valid if it has a [supported](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#testing-your-matter-device-with-apple-home) device type and its required clusters added to it. You find more information about device types and their required clusters in the matter specs. Adding these endpoints and clusters can be done using cli commands that are defined in UserInputBackend.cpp. You can type these in after launching the dynamic-chip-bridge-app.

Even if you go through these steps your added devices won't work 100% because the commands support for dynamically generated clusters is still being worked on: #23840. So maybe try checking out their progress before trying the steps above.

Apollon77, 2023-01-16:
@bndkk I saw your (deleted?) comment ... I thought that the current version of the dynamic-bridged-app is in any case having the "some sockets" by default? or did I miss anything?

bndkk, 2023-01-16:
I don't know what you mean by "some sockets".

Apollon77, 2023-01-16:
Sorry misstyped ... I meaned lights ... I refer to https://github.com/project-chip/connectedhomeip/tree/master/examples/dynamic-bridge-app/linux#bridge-implementation-example

bndkk, 2023-01-16:
yeah I wouldn't read that because that's still the README of the non-dynamic [bridge-app example](https://github.com/project-chip/connectedhomeip/tree/master/examples/bridge-app/linux. It was just copied from there, never really updated which is a problem in itself.

No there are no endpoints added to the bridge by default, you have to add them yourself.

Apollon77, 2023-01-16:
> No there are no endpoints added to the bridge by default, you have to add them yourself.

aahhhhh :-( Now it makes a lot of sense .... (and in fact the other app "bridge-app" is not building ... tried that one first and created #24443 ... Then I close here because then it already might make sense

closed 2023-01-16 by Apollon77