project-chip/connectedhomeip #24463: [Q] Java controller for non-Android environment

stlevkov, 2023-01-17:
Hi Matter Team,

I'm currently trying to use the Java Controller in a non-Android environment.
The code is successfully compiled, if i use the standard JDK Path, instead of Android-SDK. 
Using the provided example of Java-Chip-Tool i am able to run only discovery and pairing. However, there are still some imports pointing to Android, for example:
- [ChipDeviceController.java](https://github.com/project-chip/connectedhomeip/blob/master/src/controller/java/src/chip/devicecontroller/ChipDeviceController.java)
- Many of the jni files are started with prefix Android.

I'm not sure if it is okay to start the development of java controller without more information about the future of this Controller. So i have a few question and will be glad to have a short resolution which can help me with some decisions:

- Is it going to be completely Android-free and will able to be certified as well as a compliant matter controller?
- If it does, in which Matter version is expected to be released?
- Is it stable for developing a production version of Java Matter Controller?
- Do you plan to develop more functionalities for the java-chip-tool except the discovery and pairing?

Thank you in advance.

yunhanw-google, 2023-02-08:
Hi, Please consider to use 
https://github.com/project-chip/connectedhomeip/tree/master/examples/java-matter-controller, which can run java controller in linux environment to test accessary.

some instructions are as below.
-- Build
docker run -it -v /workspace/connectedhomeip:/connectedhomeip connectedhomeip/chip-build:0.6.06
git config --global --add [safe.directory](http://safe.directory/) /connectedhomeip
git config --global --add [safe.directory](http://safe.directory/) /connectedhomeip/third_party/pigweed/repo
export JAVA_PATH=/usr/lib/jvm/java-8-openjdk-amd64/
source scripts/bootstrap.sh
./scripts/build/build_examples.py --target linux-x64-java-matter-controller build
./scripts/build/build_examples.py --target linux-x64-all-clusters build

--In Terminal A
docker exec -it c20375aeb713 bash
rm -rf /tmp/chip* & ./out/linux-x64-all-clusters/chip-all-clusters-app

--In terminal B
docker exec -it c20375aeb713 bash
java -verbose:jni -Xcheck:jni -jar -Djava.library.path=out/linux-x64-java-matter-controller/lib/jni out/linux-x64-java-matter-controller/bin/java-matter-controller pairing onnetwork-long 1 20202021 3840 1000

stlevkov, 2023-02-14:
Hi, 
Thank you, @yunhanw-google ,
I have already tried that, and cannot figure out, how to send cluster commands. 
In the documentation page, that you linked, it is stated " - Send on/off cluster requests to a Matter device", but only two clusters are supported from the java-matter-controller:

- pairing
- discover

Can you give an example, of how to send on/off command from this java-controller?
Thank you in advance.