project-chip/connectedhomeip #24466: Need YAML support to set SuppressResponse to false

manjunath-grl, 2023-01-17:
### Reproduction steps / Feature

       As per test-plan in step 9: Reads AccessControlEntryChanged event even though value match's but seeing " CHIP Error 0x0000002F: Invalid argument "
Effected tests:
TC-ACL-2.8
TC-ACL-2.10
Reproduction steps:
1. Add test ACL-2.8 and run below commands.
2. Regen
'''./scripts/tools/zap_regen_all.py --type tests'''
3. Build
''' ./scripts/run_in_build_env.sh \
                     "./scripts/build/build_examples.py \
                        --target linux-x64-chip-tool-no-ble-tsan-clang \
                        --target linux-x64-all-clusters-no-ble-tsan-clang \
                        --target linux-x64-lock-no-ble-tsan-clang \
                        --target linux-x64-ota-provider-no-ble-tsan-clang \
                        --target linux-x64-ota-requestor-no-ble-tsan-clang \
                        --target linux-x64-tv-app-no-ble-tsan-clang \
                        --target linux-x64-bridge-no-ble-tsan-clang \
                        build \ "
 '''
4. Python test run:
''' ./scripts/run_in_build_env.sh \
                  "./scripts/tests/run_test_suite.py \
                     --chip-tool ./out/linux-x64-chip-tool-no-ble-tsan-clang/chip-tool \
                     run \
                     --iterations 1 \
                     --test-timeout-seconds 120 \
                     --all-clusters-app ./out/linux-x64-all-clusters-no-ble-tsan-clang/chip-all-clusters-app \
                     --lock-app ./out/linux-x64-lock-no-ble-tsan-clang/chip-lock-app \
                     --ota-provider-app ./out/linux-x64-ota-provider-no-ble-tsan-clang/chip-ota-provider-app \
                     --ota-requestor-app ./out/linux-x64-ota-requestor-no-ble-tsan-clang/chip-ota-requestor-app \
                     --tv-app ./out/linux-x64-tv-app-no-ble-tsan-clang/chip-tv-app \
                     --bridge-app ./out/linux-x64-bridge-no-ble-tsan-clang/chip-bridge-app \"
'''

### Platform

other

### Type

YAML tested

### Anything else?

Diff link:  https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:ACL_SuppressResponse_error

[image]

[ACL_2_8.zip](https://github.com/project-chip/connectedhomeip/files/10433771/ACL_2_8.zip)

manjunath-grl, 2023-01-17:
@krypton36 opening up this ticket as discussed last week

bzbarsky-apple, 2023-01-18:
@manjunath-grl Please attach the full log, not just the screenshot.  It's impossible to tell what SuppressResponse has to do with anything here (likely nothing), since what type of message we're looking at is not shown in the screenshot.

manjunath-grl, 2023-01-18:
@bzbarsky-apple I already attached a zip file named ACL_2_8.zip it contains YAML script, Execution logs and screenshot

bzbarsky-apple, 2023-01-18:
> it contains YAML script, Execution logs and screenshot

Ah, it was entirely unclear that it contained logs, given the naming.  

So what the log shows is a ReadRequest followed by a ReportData.  Since this is a read, and it's not being chunked, SuppressResponse is correctly set true by the server, so this has nothing to do with SuppressResponse and this issue's title needs to be fixed to explain what the actual problem you are seeing is.

I assume the question is why are we getting an invalid argument error?  That's coming from this code:
```
        case 10:
            switch (mTestSubStepIndex)
            {
            case 0:
                VerifyOrReturn(CheckValue("status", chip::to_underlying(status.mStatus), 0));
                {
                    chip::app::Clusters::AccessControl::Events::AccessControlEntryChanged::DecodableType value;
                    VerifyOrReturn(CheckDecodeValue(chip::app::DataModel::Decode(*data, value)));
                }
                mTestSubStepIndex++;
                break;
            default:
                LogErrorOnFailure(ContinueOnChipMainThread(CHIP_ERROR_INVALID_ARGUMENT));  <===
                break;
            }
            break;
```
because `mTestSubStepIndex` is 1.  And the test does not expect that, because the expected value for the test is wrong.  It's listing a single response that is an array of three events.  But the actual thing you get back are effectively _three responses_, each one a single event.  You want to make something like the following change:

```
diff --git a/src/app/tests/suites/certification/Test_TC_ACL_2_8.yaml b/src/app/tests/suites/certification/Test_TC_ACL_2_8.yaml
index 4bc7200fc6..119ade7ce6 100644
--- a/src/app/tests/suites/certification/Test_TC_ACL_2_8.yaml
+++ b/src/app/tests/suites/certification/Test_TC_ACL_2_8.yaml
@@ -165,7 +165,8 @@ tests:
       command: "readEvent"
       event: "AccessControlEntryChanged"
       response:
-          value:
+          - values:
+              - value:
                   {
                       AdminNodeID: null,
                       AdminPasscodeID: 0,
@@ -180,7 +181,8 @@ tests:
                           },
                       FabricIndex: TH1FabricIndex,
                   }
-          value:
+          - values:
+              - value:
                   {
                       AdminNodeID: 112233,
                       AdminPasscodeID: null,
@@ -195,7 +197,8 @@ tests:
                           },
                       FabricIndex: TH1FabricIndex,
                   }
-          value:
+          - values:
+              - value:
                    {
                       AdminNodeID: 112233,
                       AdminPasscodeID: null,
```

and similar for step 11.

In general, I recommend looking at `src/app/tests/suites/TestEvents.yaml` for examples of how to use readEvent.

closed 2023-01-18 by bzbarsky-apple