project-chip/connectedhomeip #24467: Error compiling on raspberry pi

shun-dgut, 2023-01-17:
### Reproduction steps

follow this guide
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md#installing-prerequisites-on-linux
An error occurred
[image]

### Bug prevalence

always

### GitHub hash of the SDK that was being used

bd94d060c3656

### Platform

raspi

anoop18pant, 2023-01-25:
The reason for above bug is that your submodules are not cloned correctly. Please check the thirdparty/nlfaultinjection/repo will be blank.

try to clone again.

Best of luck

shun-dgut, 2023-01-28:
> The reason for above bug is that your submodules are not cloned correctly. Please check the thirdparty/nlfaultinjection/repo will be blank.
> 
> try to clone again.
> 
> Best of luck

I've tried updating submodules, but still no files.

[image]

closed 2023-10-15 by stale[bot]