project-chip/connectedhomeip #24481: [ESP32] AC unit device: Multiple clusters per device, 3 questions

CrowdedFuzzball, 2023-01-18:
Hello!

Looking ahead - I understand that the project repository is probably not the best place to ask questions. However, the search has so far turned up no other alternative places where the implementation of the protocol can be discussed. So please excuse me if this is the wrong question.

Perhaps this thread will help someone who is looking for information like me.

What I am trying to achieve:
Create a bridged device to control the air conditioner, that is, the device must be able to change the temperature, operating mode, be able to control the blowing speed and control the swing.
Thermostat controls should be leading one, so it should be shown first.

So, to the questions:
**1. Device type**
Based on Device-Library-Specification (Version 1.0) there are 2 device types that is suitable for implementing AC unit.
There are Heating/Cooling unit (0x300) and Thermostat (0x301)
Seems like that Heating/Cooling is most suitable one. 
But when i try to test this device types in Home.app after provisioning 0x300 didn't shown in interface, but 0x301 shows as thermostat.
At the same time, in the /src/app/zap-templates/zcl/data-model/chip/matter-devices.xml file, you can see that these two devices have a different domain. XCHIP and HA. At first I thought that it affects something, but I did not find direct confirmation.

In this regard, the question is how to understand which types of devices will work with the user's end device (Google Home, Apple Home.app etc), and which ones will not?
Or should all devices be shown on all systems?

**2. Multiple clusters per device**
Since Heater/Cooler is not displayed in the interface in Home.app, we then select the Thermostat device type for implementation.
According on Device-Library-Specification (Version 1.0) Fan Control cluster is optional for that device type. 
Great, lets use it.

<details>
  <summary>Dynamic endpoint definition</summary>

```
// ---------------------------------------------------------------------------
//
// THERMOSTAT ENDPOINT: contains the following clusters:
//   - Thermostat
//   - FanControl
//   - Descriptor
//   - Bridged Device Basic

DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(thermostatAttrs)
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_LOCAL_TEMPERATURE_ATTRIBUTE_ID		, INT16S, 	2, ZAP_ATTRIBUTE_MASK(NULLABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_OCCUPIED_COOLING_SETPOINT_ATTRIBUTE_ID	, INT16S, 	2, ZAP_ATTRIBUTE_MASK(MIN_MAX) | ZAP_ATTRIBUTE_MASK(WRITABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_OCCUPIED_HEATING_SETPOINT_ATTRIBUTE_ID	, INT16S,	2, ZAP_ATTRIBUTE_MASK(MIN_MAX) | ZAP_ATTRIBUTE_MASK(WRITABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_MIN_HEAT_SETPOINT_LIMIT_ATTRIBUTE_ID	, INT16S, 	2, ZAP_ATTRIBUTE_MASK(MIN_MAX) | ZAP_ATTRIBUTE_MASK(WRITABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_MAX_HEAT_SETPOINT_LIMIT_ATTRIBUTE_ID	, INT16S, 	2, ZAP_ATTRIBUTE_MASK(MIN_MAX) | ZAP_ATTRIBUTE_MASK(WRITABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_MIN_COOL_SETPOINT_LIMIT_ATTRIBUTE_ID	, INT16S, 	2, ZAP_ATTRIBUTE_MASK(MIN_MAX) | ZAP_ATTRIBUTE_MASK(WRITABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_MAX_COOL_SETPOINT_LIMIT_ATTRIBUTE_ID	, INT16S, 	2, ZAP_ATTRIBUTE_MASK(MIN_MAX) | ZAP_ATTRIBUTE_MASK(WRITABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_MIN_SETPOINT_DEAD_BAND_ATTRIBUTE_ID	, INT8S, 	1, ZAP_ATTRIBUTE_MASK(MIN_MAX) | ZAP_ATTRIBUTE_MASK(WRITABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_CONTROL_SEQUENCE_OF_OPERATION_ATTRIBUTE_ID, ENUM8, 	1, ZAP_ATTRIBUTE_MASK(MIN_MAX) | ZAP_ATTRIBUTE_MASK(WRITABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_SYSTEM_MODE_ATTRIBUTE_ID			, ENUM8, 	1,ZAP_ATTRIBUTE_MASK(WRITABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_FEATURE_MAP_SERVER_ATTRIBUTE_ID		, BITMAP32, 4, 0),
DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(fanControlAttrs)
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_FAN_MODE_ATTRIBUTE_ID		, ENUM8, 	1, ZAP_ATTRIBUTE_MASK(MIN_MAX) | ZAP_ATTRIBUTE_MASK(WRITABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_FAN_MODE_SEQUENCE_ATTRIBUTE_ID	, ENUM8, 	1, ZAP_ATTRIBUTE_MASK(MIN_MAX) | ZAP_ATTRIBUTE_MASK(WRITABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_PERCENT_SETTING_ATTRIBUTE_ID	, INT8U, 	1, ZAP_ATTRIBUTE_MASK(MIN_MAX) | ZAP_ATTRIBUTE_MASK(WRITABLE) | ZAP_ATTRIBUTE_MASK(NULLABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_PERCENT_CURRENT_ATTRIBUTE_ID	, INT8U, 	1, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_SPEED_SETTING_ATTRIBUTE_ID	, INT8U, 	1, ZAP_ATTRIBUTE_MASK(MIN_MAX) | ZAP_ATTRIBUTE_MASK(WRITABLE) | ZAP_ATTRIBUTE_MASK(NULLABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_ROCK_SUPPORT_ATTRIBUTE_ID		, BITMAP8, 	1, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_ROCK_SETTING_ATTRIBUTE_ID		, BITMAP8, 	1, ZAP_ATTRIBUTE_MASK(WRITABLE)),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_FEATURE_MAP_SERVER_ATTRIBUTE_ID	, BITMAP32, 4, 0),
DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

constexpr chip::CommandId ThermostatIncomingCommands[] = {
	chip::app::Clusters::Thermostat::Commands::SetpointRaiseLower::Id,
	chip::kInvalidCommandId,
};

DECLARE_DYNAMIC_CLUSTER_LIST_BEGIN(bridgedThermostatClusters)
	DECLARE_DYNAMIC_CLUSTER(Thermostat::Id	, thermostatAttrs, ThermostatIncomingCommands, nullptr),
	DECLARE_DYNAMIC_CLUSTER(FanControl::Id	, fanControlAttrs, nullptr, nullptr),
	DECLARE_DYNAMIC_CLUSTER(Descriptor::Id	, descriptorAttrs, nullptr, nullptr),
	DECLARE_DYNAMIC_CLUSTER(BridgedDeviceBasic::Id, bridgedDeviceBasicAttrs, nullptr, nullptr)
DECLARE_DYNAMIC_CLUSTER_LIST_END;

// Declare Bridged Thermostat endpoint
DECLARE_DYNAMIC_ENDPOINT(bridgedThermostatEndpoint, bridgedThermostatClusters); 
```
</details>

Device definition is 

```
const EmberAfDeviceType gBridgedThermostatDeviceTypes[] 	= { { DEVICE_TYPE_THERMOSTAT, DEVICE_VERSION_DEFAULT }, { DEVICE_TYPE_BRIDGED_NODE, DEVICE_VERSION_DEFAULT } };
```

If I read the documentation correctly, then we only have to define the device, and the clusters are already described inside the device.
However (again - I test everything in Home.app) - when adding such a device, I see only a Thermostat without fan control.
If i change the device definition to:

```
const EmberAfDeviceType gBridgedThermostatDeviceTypes[] 	= { { DEVICE_TYPE_THERMOSTAT, DEVICE_VERSION_DEFAULT }, { DEVICE_TYPE_FANCONTROL, DEVICE_VERSION_DEFAULT }, { DEVICE_TYPE_BRIDGED_NODE, DEVICE_VERSION_DEFAULT } };
```

I can see Thermostat and Fan Control on one device page, but fan control is shown before Thermostat and seems like Fan control interprets as leading one. What should not be even in the order of clusters in the documentation.

In this regard, the question: could you please tell me, how to correctly set devices with several clusters? 
Is it required to define each server cluster as a device? 
Is it possible to somehow control the display order of the clusters (so that the thermostat comes first when displayed)?

**3. Plain Thermostat excursuses**

If we discard everything related to the fan controller, another problem arises.
The thermostat somehow strangely works with limits
For example, MinHeatSetpointLimit (0x0015) are defined in cluster attributes.
Value for this attribute returned by code (It is definitely returned).

```
else if ((     attributeId == ZCL_MIN_COOL_SETPOINT_LIMIT_ATTRIBUTE_ID 
          || attributeId == ZCL_MIN_HEAT_SETPOINT_LIMIT_ATTRIBUTE_ID
          || attributeId == ZCL_ABS_MIN_HEAT_SETPOINT_LIMIT_ATTRIBUTE_ID
          || attributeId == ZCL_ABS_MIN_COOL_SETPOINT_LIMIT_ATTRIBUTE_ID) 
          && (maxReadLength == 2))
{
  int16_t MinValue = 1600;
  memcpy(buffer, &MinValue, sizeof(MinValue));
}
```

But seems like thermostat in Home.app ignores this value and set 10.0C. 
This applies to other limits too.

Also i can see in log following messages
```
**chip[DMG]: Error retrieving data from clusterId: 0x0000_0201, err = b**
```

And it is interesting that sometimes these messages appear, and sometimes not. 
And for example, there may be an attempt to read the attribute unsuccessful, but then successful. 
The attributes themselves, on which an error occurs at each start, can also be different.

The read error seems to be caused in one of the following code blocks in ember-compatibility-function.cpp:

```
AttributePathIB::Builder & attributePathIBBuilder = attributeDataIBBuilder.CreatePath();
     ReturnErrorOnFailure(attributeDataIBBuilder.GetError());
```

```
    attributePathIBBuilder.Endpoint(aPath.mEndpointId)
        .Cluster(aPath.mClusterId)
        .Attribute(aPath.mAttributeId)
        .EndOfAttributePathIB();
    ReturnErrorOnFailure(attributePathIBBuilder.GetError());
```

Probably an error somewhere in my code or definition of attributes.
Tell me, please, is there any mechanism for correctly debugging the path to the attributes? 
A cursory search in the code did not allow finding one and, frankly, it is not clear where exactly to look for an attribute reading error, since they seem to be defined according to the documentation.

Thank you in advance!

bzbarsky-apple, 2023-01-18:
>  1. Device type

There is work right now to add a "Room Air Conditioner" device type.  That said, you might not be dealing with a single-room AC.

>  Seems like that Heating/Cooling is most suitable one.

That seems plausible.  Or you could implement both that and Thermostat, on different endpoints.

> But when i try to test this device types in Home.app after provisioning 0x300 didn't shown in interface

Is this in Apple Home?  Apple Home might not (yet) support that device type...  That's not a protocol issue, it's an issue with Apple Home, if so.

> you can see that these two devices have a different domain

The domain bit is not actually used for anything; it's a holdover from a different use of the XML in question.

> he question is how to understand which types of devices will work with the user's end device (Google Home, Apple Home.app etc), and which ones will not?

It depends on the implementation status in the controller.  Not all controllers will support all device types, though I suspect the goal for Google Home, Apple Home, etc is to be pretty complete eventually.

> According on Device-Library-Specification (Version 1.0) Fan Control cluster is optional for that device type.
> Great, lets use it.

A few things:

1. It's not just optional, it's also provisional ("P").  Which means that it should not be used yet, until it stops being provisional.

    And it's provisional because there are no certification tests for it, so it can't be certified yet.
2. For the Thermostat device type what's optional is a Fan Control _client_.  As in, the thermostat can optionally control some fan.  But the attached dynamic endpoint definition is defining a Fan Control _server_.
3. Using `DEVICE_VERSION_DEFAULT` here is probably wrong.  I expect you have `DEVICE_VERSION_DEFAULT` set to 1, but you are trying to implement version 2 of the Thermostat device type, I would think (version 1 is Zigbee-only).

> However (again - I test everything in Home.app) - when adding such a device, I see only a Thermostat without fan control.

This is correct, since a Thermostat is not supposed to have a Fan Control _server_.

>  If i change the device definition to:

You're not supposed to have multiple unrelated application device types on the same endpoint.  From "9.2. Endpoint Composition" in the Matter spec:

> Each simple endpoint SHALL support only one Application device type with these exceptions:
> * The endpoint MAY support additional device types which are subsets of the Application device type (the superset).
> * The endpoint MAY support additional device types (application, utility or node device types) as defined by each additional device type.

> In this regard, the question: could you please tell me, how to correctly set devices with several clusters?

I'm not sure whether the above answers this question, but in brief: you pick an application device type (or family of them if you are using the "subsets" exception), then you implement the _server_ clusters for that device type and those will show up on the controller side.  At least if everything is working as it should.

>  Is it required to define each server cluster as a device?

"No", if I understand the question right, but I might be mis-understanding it.

>  Is it possible to somehow control the display order of the clusters (so that the thermostat comes first when displayed)?

Not relevant in this case, since there is no device type that has both a Thermostat server and a Fan Control server.

> But seems like thermostat in Home.app ignores this value and set 10.0C.

It's hard to say anything intelligent here without logs and whatnot.  You might want to report the Home issue following directions similar to https://github.com/project-chip/connectedhomeip/issues/23368#issuecomment-1293939732 with logs from the device side, a sysdiagnose from the Home.app side, etc.

> **chip[DMG]: Error retrieving data from clusterId: 0x0000_0201, err = b**

That _may_ be normal.  0x0b is CHIP_ERROR_NO_MEMORY, and if this is a wildcard read (which gets chunked because it's larger than a single MTU) that could indicate our chunk filled up and we need to send it then try reading that attribute again in the next chunk.  Again, the full log would make it clearer what's going on.

> And it is interesting that sometimes these messages appear, and sometimes not.

It would depend on the exact state of the device and the client, etc...

> The read error seems to be caused in one of the following code blocks in ember-compatibility-function.cpp:

Yes, that's trying to encode into the buffer.  If it runs out of buffer, it will return an error.

Hopefully some of the above helps a bit.

bzbarsky-apple, 2023-01-19:
>  As logic tells me, all devices in the subset should simply return the same subset ID, but how exactly and what rules are not yet clear.

I'm not sure what you mean by "subset ID".  The "subset" thing I mentioned was when an endpoint has both a "On/Off Light" device type and a "Color Light" device type, say.  The former is a subset of the latter, and is there just for compatibility with controllers that don't know how to do Color Light but can handle an On/Off light.

CrowdedFuzzball, 2023-01-19:
@bzbarsky-apple i'm sorry, i meant Descriptor Cluster
But first I have to carefully study the documentation on this.

CrowdedFuzzball, 2023-01-26:
@bzbarsky-apple 
It is very good that the protocol has documentation and it is so comprehensive!
There were far fewer questions.

Thanks for your answers

Perhaps I will ask only one question - is there some kind of testing mechanism in order to determine that all cluster attributes are set correctly (types, return values, etc)?
Conditionally, a tool that can be launched and which checks the compliance of the created device with the protocol?

I initially thought that chip-tool would help here, but so far it looks like it's not.

bzbarsky-apple, 2023-01-26:
> is there some kind of testing mechanism in order to determine that all cluster attributes are set correctly (types, return values, etc)?

Apart from the certification tests, you mean?

Some of those are automated.  See https://github.com/project-chip/connectedhomeip/blob/master/.github/workflows/tests.yaml#L219-L231

You can pass a `--pics-file` argument to `scripts/tests/run_test_suite.py run` (where `--iterations 1` is being passed) that represents your actual device capabilities, so unrelated tests get skipped, and point the relevant `...-app` arguments to your app as needed.

CrowdedFuzzball, 2023-01-27:
Thanks for the detailed answers
I think that now I have enough information and / or I know where to look if necessary.

Thank you for your work!

closed 2023-01-27 by CrowdedFuzzball