project-chip/connectedhomeip #24486: Decide whether Python APIs should preserve acronyms (in some names?)

bzbarsky-apple, 2023-01-18:
We have enum items named things like "OK" which turn into "kOk" in the python APIs (so we get `Enums.NodeOperationalCertStatusEnum.kOk`). Should this be "kOK"? What about other acronyms (ACL, USB, URL) that currently end up getting turned in Acl, Usb, Url, respectively?

Need to decide what this API surface should look like.

@tehampson @mrjerryjohns @tcarmelveilleux @yunhanw-google @yufengwangca

yunhanw-google, 2023-01-19:
https://docs.python.org/3/library/enum.html#creating-an-enum
it seems the official prefers the uppercase without k

bzbarsky-apple, 2023-01-20:
I suspect the "k" was carried over from C++, where it was added because some enum names (incorrectly) started with a number.  I'm pretty sure at this point all enums don't do that.

closed 2023-10-15 by stale[bot]