project-chip/connectedhomeip #24513: [Platform] ZAP not outputting .matter file

step0035, 2023-01-19:
### Reproduction steps / Feature

not sure if this is the correct way of generating a .matter file, here is the command that i ran.

```
python3 ./scripts/tools/zap/generate.py examples/all-clusters-app/all-clusters-common/all-clusters-app.zap -t src/app/zap-templates/matter-idl.json -o zzz_generated/all-clusters-app/zap-generated/
```

nothing was outputted to `zzz_generated/all-clusters-app/zap-generated/`

zap version: `v2023.01.09-nightly`

### Platform

other

### Type

Manually tested with SDK

bzbarsky-apple, 2023-01-19:
The .matter file is copied over to `examples/all-clusters-app/all-clusters-common/all-clusters-app.matter` by the command listed above, no?  See https://github.com/project-chip/connectedhomeip/blob/8ad1073c4dd7d76b2ee89f86ce2f7c86865e7ae4/scripts/tools/zap/generate.py#L161-L177

step0035, 2023-01-20:
verified it is copied to `examples/all-clusters-app/all-clusters-common/all-clusters-app.matter`, thanks

closed 2023-01-20 by step0035