project-chip/connectedhomeip #24571: Remove stale bot

ShadowJonathan, 2023-01-21:
I want to ask to remove the stale bot from this repo, it will not serve any comprehensive purpose for the long term of this project;
- It'll close issues that haven't had any "activity", many bugs need to wait for maintainers to address them, and so there's not much "activity" to be had, and people talking and bumping them does not count as activity, only noise
- As such, it could and will close issues which have been/will be plaguing matter for a long time, and together with the above, will effectively be dismissed and ignored "automatically" by this bot
- Issue count is not a metric. Issue count is not a metric. Issue count. is. not. a. metric.
  - Even if it is a metric, it should not be target, at the detriment of the project
- Any concerns about why a stale bot should exist can better be addressed by solving other problems:
  - Not enough resources to triage support questions? Front-load knowledge, create wiki resources, and generally point people to them
  - Not enough resources to triage issues in general? That is a bigger problem, not solved by shotgunning untriaged (or even post-triage(!)) issues
  - Too many duplicate issues? Mark as duplicate, find or make tracker issues and discuss the core problems, cultivate.

There is no excuse for a stale bot to be in this repo.

bzbarsky-apple, 2023-01-23:
@andy31415 @woody-apple

ShadowJonathan, 2023-08-07:
No.

ShadowJonathan, 2023-08-07:
@andy31415 @woody-apple @bzbarsky-apple

andy31415, 2023-08-09:
I am torn on this one  - when we have a lot of issues open, old ones tend to be not looked at (seems to happen the same with PRs unfortunately). As a result, only issues with traffic (even if not that informative "still happening in the latest master SHA) actually get looked at.

In my mind issue count is absolutely not a metric, however we also need some way to assess what issues are still happening and important. As a developer it can be frustrating to look to fix an issue only to find out it is not reproducible anymore (or wrongly assume it is not by testing just a few times however test environment having changed).

Any suggestions on approaches? Ideally I would want a smarter bot - clear issue reports to be kept open while actually potentially stale issues (vague reproduction, people having looked at them and never again) to actually get obsoleted.

ShadowJonathan, 2023-08-09:
That might be a good compromise, but please make sure your issue triage process makes sure that no issues fall through the cracks.

For instance, issues that haven't gotten to triage yet. That situation is understandable when developers are overworked and can't handle everything, but issues shouldn't languish, it should be FIFO. Issues (imo) may be gotten stale if the response depends on the issuer, such as when devs are waiting for better reproduction info, and if a few months pass by without the author getting to it, it almost certainly has fallen through.

However, when issues are confirmed reproduced, they should get labels that would prevent stale bots from reaping them, and a developer could go through and try to reproduce it again, to see if the issue still exists.

But only through that way should an well-defined issue in progress get closed for staleness; manually.