project-chip/connectedhomeip #24576: [Feature] [ESP32] Support esp-idf v5.0 release

shubhamdp, 2023-01-23:
### Feature description

[ESP-IDF Release v5.0](https://github.com/espressif/esp-idf/releases/tag/v5.0) is out some time back and it is not supported in the SDK.

This issue is to track the progress for supporting esp-idf v5.0 alongside esp-idf v4.4.3

### Platform

esp32

vindicatorr, 2023-01-25:
Just adding myself to this. I thought I was going crazy because I thought I had commented on this sort of thing, and I was looking for references of myself as author. Only did I finally see my comment as "commenter" (https://github.com/project-chip/connectedhomeip/issues/17239#issuecomment-1368643196)

I only mention it again because I just encountered it again regarding the error "format \<...\> expects argument of type \<...\>, but argument \<#\> has type '\<type\>' {aka ''\<definition\>'}". I've got to put that in my notes so I don't have to go checking the function of my mind.

dhrishi, 2023-02-08:
@shubhamdp Please update the apps which are now supported. If everything is complete, close the issue.

shubhamdp, 2023-02-08:
@dhrishi updated the supported apps. I will put up one more PR to support remaining apps and once that is merged, I'll close the issue

closed 2023-02-13 by bzbarsky-apple