project-chip/connectedhomeip #24577: [BUG] Apple iOS 16.2 showing Matter Bridge Devices with status Accessory is not responding

stlevkov, 2023-01-23:
### Reproduction steps

Prerequisite:
1. Linux 22.04 VM already connected to local network via WiFi
2. iPhone with 16.2 update
3. CHIP SDK with v1.0.0.2

Reproduce steps:
1. Compile bridge-app example (non-dynamic)
3. Remove any previously saved session data ``` rm -rf /tmp/chip* ```
4. Start bridge-app example
5. Open Home App on iPhone
6. Add Accessory
7. Commission successfully
8. Identify all of the Matter Devices found manually (without identify command)
9. All of the Matter Accessory shown with status "_This accessory is not responding_"

Error found in bridge-app logs:
```
[1674459250.087532][16500:16500] CHIP:DL: wpa_supplicant: connected to network
(process:16500): GLib-GIO-CRITICAL **: 09:34:10.087: g_dbus_proxy_call_sync_internal: assertion 'G_IS_DBUS_PROXY (proxy)' failed
[1674459250.087632][16500:16500] CHIP:DL: wpa_supplicant: failed to save config: unknown error
```

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

4088a77f557e8571a39338fad51a1d8eb0131d79

### Platform

darwin

### Platform Version(s)

DUT bridge-app on Linux 22.04

### Anything else?

Hi, 
I have started bridge implementation with dynamic endpoints. Everything is working fine with the linux chip-tool, but i am not able to present my devices to the Apple HomeApp. Get the same status as with the bridge-example-app: **This accessory is not responding**.
There are no errors in the log affecting the commissioning with my implementation. I found some error during the commissioning with the bridge-example-app.

bzbarsky-apple, 2023-01-23:
@stlevkov Can you please follow instructions below to file a feedback assistant ticket? 

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

sumittiwari4527, 2023-01-30:
@stlevkov  any update on this issue, did you find any solution as I am facing the same issue for my device.

stlevkov, 2023-02-03:
Hi,
I was able to initialize the bridge-app successfully with the AppleHome, but using the chip-tool:

-   first commission the bridge-app with the Chip-tool
-   send open-commissionining-window command
-   use the newly generated code for pairing

However, I'm still not able to pair it directly. I will try to upload the required logs to the feedback assistant ticket.

stlevkov, 2023-02-08:
@bzbarsky-apple, attached the logs in fb: 11984867

bzbarsky-apple, 2023-02-09:
@stlevkov Thank you for attaching that!  What I see in there is an iPhone sysdiagnose and the log from the bridge app.  Is there also an AppleTV or HomePod involved?

stlevkov, 2023-02-10:
@bzbarsky-apple I have an HomePod only, but I am not sure how to attach logs from it.

bzbarsky-apple, 2023-02-10:
@stlevkov The directions and profile for "HomeKit for HomePod" at https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=other might let you do that?

stlevkov, 2023-02-14:
@bzbarsky-apple I attached the sysdiagnose logs from the HomePod Mini.

stlevkov, 2023-02-14:
@bzbarsky-apple Yes, I performed clean reproduction:

- Reset HomePod Mini
- Setup HomePod Mini
- Install Log Profile
- Perform pairing with Bridge App
- Bridge App Accessory paired, but show status 'not responding'
- Export sysdiagnose logs from the button Export

I noticed the following:
My Bridge App is already connected to WiFi network - no BLE support.

### Bridge-app-example 
- Pairing the bridge-app-example is OK, after first commission it with chip-tool, and send open-commissioning-window.
- Pairing with HomeApp - The Matter accessory is displayed in HomeApp as Bridge and its working as expected.

### My Bridge-app
- I don't have support for open-commissioning-window, that's why I directly pair it with HomeApp
- After pairing is complete - The Matter accessory is displayed in HomeApp not as Bridge, and status 'not responding'
- I can see all of my endpoints initialized with their Device Types, but with status 'No Response', summarized under Matter Accessory
- After commission is complete, There is no matter messages attempts in the Bridge App logs
- Restart the iPhone, open HomeApp - again, status 'Updating' then 'No response', opening the accessory shows 'not responding'

Observing the differences showing that, the bridge-app-example has:
- EP0, Root Node, partsList - all of the Endpoints, including 1
- EP1, Aggregator, partsList - all of the Endpoints, not including 0 and 1

My bridge app has:
- EP0, Root Node, Aggregator, partsList - all of the Endpoints, starting from 2. I am using the Root Node as Aggregator as well.

Is this could be the reason for not initializing the accessory?

bzbarsky-apple, 2023-02-16:
> Export sysdiagnose logs from the button Export

Weird... It's just not showing any Matter activity on that homepod.

> Is this could be the reason for not initializing the accessory?

It's possible... @nivi-apple ?

stlevkov, 2023-02-17:
Hi,

Implementing the Aggregator Endpoint as separate from the Root Node endpoint - fixed the problem.

per 1.0 core spec:
**9.12.2. Exposing functionality and metadata of Bridged Devices**
_Additionally, it SHALL expose an endpoint with the device type Aggregator which has a Descriptor
cluster with a PartsList attribute containing all the endpoints representing those Bridged
Devices. See Endpoint Composition for the concept of hierarchical composition._

So it seems like it is mandatory for the Aggregator to be a separate Endpoint provided by the Bridge.

closed 2023-02-17 by stlevkov

bzbarsky-apple, 2023-02-17:
Ah, so the spec for the Root Node device type says:

> Other non-Node device types and Application clusters SHALL NOT be supported on the same endpoint as this device type.

Aggregator is of class "Endpoint", not "Node", so cannot be on endpoint 0, period.