project-chip/connectedhomeip #24593: [BUG] ESP32/M5 board crash loops - assert failed: xQueueTakeMutexRecursive queue.c:717 (pxMutex)

kean-apple, 2023-01-23:
### Reproduction steps

I flashed my M5 board with Matter SHA - bba4182d61d56510f628307aff9844b75fb27b0d. I'm seeing the board crash loop 100% after flashing the board.

I'm using ESP IDF 3.3.2. 

Note: This does not happen with Matter SHA 9fbcf1cef8c4a42761a7fc23588ca71d75904ff5

assert failed: xQueueTakeMutexRecursive queue.c:717 (pxMutex)

Backtrace:0x40081e72:0x3ffe9c000x40092639:0x3ffe9c20 0x40098ea1:0x3ffe9c40 0x400936ee:0x3ffe9d60 0x401067c5:0x3ffe9d80 0x401069db:0x3ffe9da0 0x400f6566:0x3ffe9dc0 0x400f3375:0x3ffe9de0 0x400f342a:0x3ffe9e00 0x400d7e05:0x3ffe9e40 0x40101edc:0x3ffe9e70 0x40101f59:0x3ffe9ee0 0x400f76b1:0x3ffe9f10 0x400f8b4d:0x3ffe9f40 0x400e1698:0x3ffe9f60 0x400da9dd:0x3ffe9f90 0x400daa0e:0x3ffe9fb0 0x400dc995:0x3ffe9fd0 0x400d9ed3:0x3ffe9ff0 0x400d9b8d:0x3ffea010 0x400d8298:0x3ffea110 0x400d76b2:0x3ffea130 0x40140d5d:0x3ffea150 0x40141046:0x3ffea170 0x40141075:0x3ffea1e0 0x400958d5:0x3ffea200 
0x40081e72: panic_abort at /Users/keanlim/tools/esp-idf/components/esp_system/panic.c:402

[M5-panic-loop.txt](https://github.com/project-chip/connectedhomeip/files/10482824/M5-panic-loop.txt)

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

bba4182d61d56510f628307aff9844b75fb27b0d

### Platform

esp32

### Platform Version(s)

ESP IDF 3.3.2

bzbarsky-apple, 2023-01-23:
@shubhamdp @dhrishi

shubhamdp, 2023-01-24:
Hi @kean-apple can you confirm that you are really using IDF v3.3.2?

shubhamdp, 2023-01-24:
You should be using IDF v4.4.3.

kean-apple, 2023-01-24:
Updated to ESP-IDF 4.4.3 and using SHA 5400de1cf831c047e2bb1618e2878861fc0186b9 and the crash still happens..

[M5-crash-ESP-IDF-4-4-3.txt](https://github.com/project-chip/connectedhomeip/files/10485690/M5-crash-ESP-IDF-4-4-3.txt)

mlepage-google, 2023-01-24:
I have been seeing this too, SHA ec031378c660fee6650c61c10da3ff0388244e47
I'm on ESP-IDF v4.4.2.

PR #24607 (2023-01-24): Fix the boot loop crash on M5Stack
This is the regression from #24547, which is accessing the attributes even before Server is ready. Moved the lock initialization after service is initialized.

Fixes #24593

#### Tests
No boot loop and was able to commission the M5 stack.
closed 2023-01-24 by Damian-Nordic