project-chip/connectedhomeip #24604: [BUG] "mbedtls_pkcs5_pbkdf2_hmac" Deprecated

vindicatorr, 2023-01-24:
### Reproduction steps

Build anything that uses mbedtls for esp32.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

Master

### Platform

esp32

### Platform Version(s)

Master

### Anything else?

```
error: 'mbedtls_pkcs5_pbkdf2_hmac' was not declared in this scope; did you mean 'mbedtls_pkcs5_pbkdf2_hmac_ext'?
```

Line of the issue: https://github.com/project-chip/connectedhomeip/blob/master/src/crypto/CHIPCryptoPALmbedTLS.cpp#L406

Discussion of deprecation of `mbedtls_pkcs5_pbkdf2_hmac` being replaced with `mbedtls_pkcs5_pbkdf2_hmac_ext`: https://github.com/Mbed-TLS/mbedtls/issues/6145

Line of header containing deprecated function declaration: https://github.com/espressif/mbedtls/blob/e39975969d548572a39875ad29abd88b23285f94/include/mbedtls/pkcs5.h#L94

Line where esp32 disallows use of deprecated things: https://github.com/espressif/esp-idf/blob/49551cc48cb3cdd5563059028749616de313f0ec/components/mbedtls/port/include/mbedtls/esp_config.h#L2958

Move to using `mbedtls_pkcs5_pbkdf2_hmac_ext` instead.

jadhavrohit924, 2023-02-21:
Hi @vindicatorr, `mbedtls_pkcs5_pbkdf2_hmac` is not deprecated in idf v4.4.3 and v5.0. Thats why esp32 using `mbedtls_pkcs5_pbkdf2_hmac`. `mbedtls_pkcs5_pbkdf2_hmac_ext` will be supported in upcoming release

vindicatorr, 2023-02-21:
That's why I mentioned I was using the Master branch for everything, and it's just a heads-up for what's coming down the pipeline.

closed 2023-02-22 by andy31415