project-chip/connectedhomeip #24609: [BUG] Build error - missing pigweed libc++.a (M1 Ventura 13.1)

zcduthie, 2023-01-24:
### Reproduction steps

Attempting to build both all-clusters-app for darwin arm64 and ./gn_build.sh produce the same error.
```clang: error: no such file or directory: '../../.environment/cipd/packages/pigweed/lib/libc++.a'```
(full logs attached below)

Fresh checkout/setup and:

```
cd ~/connectedhomeip
source scripts/activate.sh
./gn_build.sh
```
[gn_build.log](https://github.com/project-chip/connectedhomeip/files/10487872/gn_build.log)

This does not occur when targeting a different (non-mac) architecture. Can _successfully_ build Android CHIP Tool via `./scripts/build/build_examples.py --target android-arm64-chip-tool build`)

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

ca086f2

### Platform

darwin

### Platform Version(s)

Apple M1, Ventura 13.1

### Anything else?

The entire `lib` folder is missing from `.environment/cipd/packages/pigweed/`

Various attempts at different versions of the SDK, pigweed, force removing `.environment` and re-bootstrapping haven't been able to fix / generate it.

[pw_doctor.log](https://github.com/project-chip/connectedhomeip/files/10487908/pw_doctor.log)

bzbarsky-apple, 2023-01-24:
Pigweed does not provide an ARM toolchain for M1/M2 macs, as far as I can tell.

The default "./gn_build.sh" tries to build things with multiple compilers, including this non-existent toolchain, for some of the tests it runs.

I think passing `enable_host_clang_boringssl_crypto_tests=false` to `./gn_build.sh` might help work around that...

@andy31415

zcduthie, 2023-01-30:
Thanks - looks like that must be the culprit.

I originally came across this issue following the instructions at [https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/linux](https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/linux):
> To compile on an Arm Mac, run:
> `$ ./scripts/run_in_build_env.sh "./scripts/build/build_examples.py --target darwin-arm64-all-clusters-no-ble-asan-clang build"`

Does this mean darwin-arm64 targeted scripts won't currently work?

bzbarsky-apple, 2023-01-30:
This should work:
> `$ ./scripts/run_in_build_env.sh "./scripts/build/build_examples.py --target darwin-arm64-all-clusters-no-ble-asan build"`

(i.e. without the "-clang" suffix).  That suffix means "use pigweed clang instead of the system one" on Mac, as far as I can tell, and on M1 there is no pigweed clang.....

bzbarsky-apple, 2023-01-30:
(And we should either fix the script to work or fix the documentation or both....)

zcduthie, 2023-02-01:
> $ ./scripts/run_in_build_env.sh "./scripts/build/build_examples.py --target darwin-arm64-all-clusters-no-ble-asan build"

Can confirm that this worked for me.

bzbarsky-apple, 2023-02-01:
I created https://github.com/project-chip/connectedhomeip/pull/24775 to fix the documentation for now.

closed 2023-02-10 by bzbarsky-apple