project-chip/connectedhomeip #24621: [BUG] Raspberry Pi 4 - zap-cli erroring out during build_python_device script

SpencerWF, 2023-01-24:
### Reproduction steps

1) Create SD card with latest Raspberry Pi aarch64 OS.
2) Run through setup steps from "https://community.arm.com/arm-community-blogs/b/internet-of-things-blog/posts/build-a-matter-home-automation-service-using-raspberry-pi-arm-virtual-hardware-and-python?CommentId=394f9717-20c8-451c-9e96-1952dbf0b905" until step 9 ("./scripts/build_python_device.sh --chip_detail_logging true")
3) Install zap-cli on raspberry pi from (https://github.com/Grunny/zap-cli)
4) Install zap from 
5) run step 9 "./scripts/build_python_device.sh --chip_detail_logging true"
6) Get error "subprocess.CalledProcessError: Command '['zap-cli', '--version']' returned non-zero exit status.

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

fc4281c356e97819f11ce05e82556d408c5a2dae

### Platform

raspi

### Anything else?

First time trying to setup, so I might just be using the wrong version of the repo.

bzbarsky-apple, 2023-01-24:
@andy31415

andy31415, 2023-01-24:
zap-cli relates to a tool in https://github.com/project-chip/zap

`Grunny/zap-cli` is most certainly not the thing you want

andy31415, 2023-01-24:
Our error messages already say:

```
You may need to install zap. Please ensure one of these applies:

- `zap-cli` is in $PATH. Install from https://github.com/project-chip/zap/releases'
...
```