project-chip/connectedhomeip #24638: [BUG] Test_TC_OO_2_3 failure

s07641069, 2023-01-25:
### Reproduction steps

@Jerry-ESP @bzbarsky-apple @chshu @woody-apple @jmar

Test_TC_OO_2_3 starts to fail after merge of PR: https://github.com/project-chip/connectedhomeip/pull/21569

I created two temporary branches where added Test_TC_OO_2_3 into GitHub CI automated tests list:

1) Just enabled run of Test_TC_OO_2_3:
Result Test_TC_OO_2_3 fail: https://github.com/s07641069/connectedhomeip/actions/runs/3995339817/jobs/6854099416
Branch: https://github.com/s07641069/connectedhomeip/commits/telink_enable_Test_TC_OO_2_3

2) Enabled run of Test_TC_OO_2_3 and reverted PR https://github.com/project-chip/connectedhomeip/pull/21569 changes:
Result Test_TC_OO_2_3 pass: https://github.com/s07641069/connectedhomeip/actions/runs/4004128687/jobs/6872889075
Branch: https://github.com/s07641069/connectedhomeip/commits/telink_fix_Test_TC_OO_2_3

Mainly problem is on change MoveToLevel command into MoveToLevelWithOnOff here:
```
        // ...else if newValue is OnOff::Commands::Off::Id...
        // "Move CurrentLevel to the minimum level allowed for the device over the
        // time period OnOffTransitionTime."
        if (useOnLevel)
        {
            // If OnLevel is defined, don't revert to stored level.
            moveToLevelHandler(endpoint, Commands::MoveToLevelWithOnOff::Id, minimumLevelAllowedForTheDevice, transitionTime,
                               chip::NullOptional, chip::NullOptional, INVALID_STORED_LEVEL);
        }
        else
        {
            // If OnLevel is not defined, set the CurrentLevel to the stored level.
            moveToLevelHandler(endpoint, Commands::MoveToLevelWithOnOff::Id, minimumLevelAllowedForTheDevice, transitionTime,
                               chip::NullOptional, chip::NullOptional, temporaryCurrentLevelCache.Value());
        }
```

Because inside moveToLevelHandler function attribute GlobalSceneControl will be set again into true, but in test expected still false.
```
    if (commandId == Commands::MoveToLevelWithOnOff::Id)
    {
        uint32_t featureMap;
        if (Attributes::FeatureMap::Get(endpoint, &featureMap) == EMBER_ZCL_STATUS_SUCCESS &&
            READBITS(featureMap, EMBER_AF_LEVEL_CONTROL_FEATURE_LIGHTING))
        {
            OnOff::Attributes::GlobalSceneControl::Set(endpoint, true);
        }
    }
```

Just to change commands back into MoveToLevel will crash new algorithm.
Needs to revert changes from PR fully or rework PR with testing Test_TC_OO_2_3.

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

612fc1c456a0c515289bab1420402d4bb149b394

### Platform

core

bzbarsky-apple, 2023-01-25:
I still don't think https://github.com/project-chip/connectedhomeip/pull/21569 was the right change to start with, and this is evidence that it's not...  @Jerry-ESP @jmartinez-silabs please take a look?

jmartinez-silabs, 2023-01-25:
There was a lot of discussion on that PR and on slack. At the time we came to an agreement and tests and cert were done using that code I am surprised this comes out now. I will revisit Jerry's PR, try the test out again an go through the code. There are a lot of corner cases in those cluster interactions

PR #24800 (2023-02-02): Fix OnOff GlobalSceneControl incorrect state change 
This PR Fixes #24638
MoveToLevelWithOnOff should only set the GlobalSceneControl to true if it produced a On action when the on off feature map cluster indicate the lighting feature.

The wrong cluster's featuremap was also verified. The command is from the level cluster but the GlobalSceneControl attributes is enable with to the onoff ligthing featuremap.

Tested using the lighting app on efr32 and the chip-tool tests Test_TC_OO_2_3
`[1675282792.995858][398020:398025] CHIP:TOO:  **** Test Complete: Test_TC_OO_2_3`
closed 2023-02-02 by jmartinez-silabs