project-chip/connectedhomeip #24645: [BUG] Control state objects in LevelControl cluster for dynamic endpoints are never initialized

raven-worx, 2023-01-25:
### Reproduction steps

I am implementing a dynamic-bridge Matter device. Some of the dynamic endpoints have a LevelControl cluster added.

The problem now is that the LevelControl cluster doesn't work when the value is about to be changed via a command.
After debugging the LevelControl cluster implementation i found that the retrieved state pointer (eg. inside `moveToLevelHandler()` method) is not initialized properly.

This is only the case for dynamically added endpoints. The dummy endpoint's state object gets initialized in `emberAfLevelControlClusterServerInitCallback()`. Especially the `maxLevel`/`minLevel` members remain 0 and thus the command never executes correctly.

The attributes are set properly, see my endpoint dump attached.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

43dd327f1cf497cdf8c90dfb2e481dd567abfaa9

### Platform

other

### Platform Version(s)

1.0.0.2

### Anything else?

[repl_dump.txt](https://github.com/project-chip/connectedhomeip/files/10501066/repl_dump.txt)

bzbarsky-apple, 2023-01-25:
@jmartinez-silabs

jmartinez-silabs, 2023-01-31:
@raven-worx Reading the issue my first thought was that the `stateTable` size didn't handle the possible dynamic endpoint count. But I just went through the code and it does. It should be able to initialize the dynamically added endpoint through the `emberAfLevelControlClusterServerInitCallback` call.

I don't have a sample app with dynamic endpoint creation to this out. 
can you let me know what is the value of `CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT` in your app, what endpoint value is dynamically created? 

Did you debug the calls of `emberAfLevelControlClusterServerInitCallback` What are the values of the endpoint parameters for each time emberAfLevelControlClusterServerInitCallbacke should match your dynamically created endpoint

raven-worx, 2023-01-31:
@jmartinez-silabs 
my `CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT` is set to 200 at the moment.

Yes i also debugged `emberAfLevelControlClusterServerInitCallback` and it is only called on startup for the fixed endpoints - which in my case is only the bridge dummy endpoint i disable later on.

It should get called when i call `emberAfSetDynamicEndpoint` to add my dynamic endpoint, i guess.

For now as a workaround i added the few lines of `emberAfLevelControlClusterServerInitCallback` to initialize the bounds attributes to the `moveToLevelHandler` and works for me so far.

bzbarsky-apple, 2023-01-31:
> It should get called when i call emberAfSetDynamicEndpoint to add my dynamic endpoint, i guess.

Do you say that in your endpoint config for the dynamic endpoint?  I am guessing not, because `DECLARE_DYNAMIC_CLUSTER` does not list the init method (just uses NULL for the cluster functions)....

bzbarsky-apple, 2023-01-31:
Ideally, we would auto-infer the init method somehow....

raven-worx, 2023-01-31:
>  I am guessing not, because `DECLARE_DYNAMIC_CLUSTER` does not list the init method (just uses NULL for the cluster functions)....

i am using the dynamic-bridge-app as a base

bzbarsky-apple, 2023-02-01:
Then that's possibly a bug in dynamic-bridge-app, depending on how it's setting up its clusters.