project-chip/connectedhomeip #24684: [BUG] onnetwork commissioning doesn't add network to networks list

pankore, 2023-01-27:
### Reproduction steps

Connect device to the same wifi network as the commissioner beforehand
Commission the device using onnetwork pairing
`./chip-tool pairing onnetwork 1 20202021`
Read the networks list, it should show 0 entries
`./chip-tool networkcommissioning read networks 1 0`

- Controller/commissioner has no way to get the wifi password
- `AddOrUpdateWiFiNetwork` is not called
- The wifi network is not added to the networks list, might lead to not being persisted on some platforms

Is this an expected behaviour? Should platform code implement adding the network or should it be the controller's job? If it is controller's job, how would the wifi password be known to the controller? 

### Bug prevalence

always

### GitHub hash of the SDK that was being used

bdefbcb39b115f7be9c30df9e106ac9302f49033

### Platform

ameba, other

bzbarsky-apple, 2023-01-27:
If a device is placed on the network through non-Matter means, then either the network should not appear in the networks list, or whatever did the placing on the network should explicitly add the network there.

The first of those is argued for by this part of the spec:

> The order of list items SHALL only be modified by the AddOrUpdateThreadNetwork, AddOrUpdateWiFiNetwork and ReorderNetwork commands. In other words, the list SHALL be stable over time, unless mutated externally.

but the spec does have provisions for the Ethernet version of the network commissioning cluster to automatically populate the list with the one entry....

So really, this needs to be raised as a spec issue to clarify behavior.

ksperling-apple, 2023-01-27:
To my reading that spec paragraph simply talks about the fact that the list *order* reflects configuration precedence and doesn't change by itself (e.g. when the primary network is unavailable and the device falls back to the secondary network).

pankore, 2023-02-16:
closing issue, onnetwork devices should not have network commissioning cluster

closed 2023-02-16 by pankore