project-chip/connectedhomeip #24687: [BUG] Unable to Commission over ble-thread using nrf52840dk kit.

SiddMishra20, 2023-01-27:
### Reproduction steps

We have been trying to commission over ble and operations over Thread. 

1. nrf52840 kit has been used as a Matter Device (MatterLight)
2. We are using a Gateway as an OTBR. 
3. Running the Chip tool over Linux PC with Ubuntu 20.04

Whenever we run the Chip-tool, it goes forward till certain steps of commissioning but later fails every time. 

### Bug prevalence

Frequent

### GitHub hash of the SDK that was being used

v1.0-branch

### Platform

nrf

### Platform Version(s)

v1.0-branch

### Anything else?

Please find attached the Log files for the same. 
Chip-Tool in Linux PC
[BugReport_Linux.log](https://github.com/project-chip/connectedhomeip/files/10516125/BugReport_Linux.log)
Connectivity Logs with NRF Kit
[BugReport_Nrf.log](https://github.com/project-chip/connectedhomeip/files/10516127/BugReport_Nrf.log)

bzbarsky-apple, 2023-01-27:
So the device gets the ConnectNetwork command:
```
D: 185490 [DMG]Received command for Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0006
```
and starts advertising operationally, in theory
```
I: 187118 [DIS]Advertise operational node 9976CA30B074B9EF-0000000000000011
```
Then the client fails operational discovery and closes the BLE session as part of its cleanup:
```
I: 202292 [DL]BLE GAP connection terminated (reason 0x16)
```
but then some time later the server is still messing with its Thread network setup:
```
D: 205462 [DL]OpenThread State Changed (Flags: 0x00000064)
D: 205467 [DL]   Device Role: ROUTER
```

The basic problem seems to be that the client is timing out operational discovery after 15 seconds (it's 45 on tip, not 15) and the server needing more than 15 seconds after it claims success from ConnectNetwork to actually be on the network.

This seems like it might be due to router promotion on the server?  See https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/5700

@Damian-Nordic

SiddMishra20, 2023-01-27:
Thanks for that Detailed analogy Boris! 
Somehow, The link shows "Page Not Found".
Is there any change I should make in my repo to increase this timeout?

Damian-Nordic, 2023-01-27:
Can you provide more details on your setup?
1. Which device you run the OTBR on?
2. How frequently does this happen?
3. Is the device connected directly to the OTBR or you use a more complex topology?
4. Would you be able to test if changing `ConnectivityMgr().SetThreadDeviceType(ConnectivityManager::kThreadDeviceType_Router);` line in the sample's `AppTask.cpp` to `ConnectivityMgr().SetThreadDeviceType(ConnectivityManager::kThreadDeviceType_FullEndDevice);` helps resolve the issue?

If the 4th point does help, it would suggest that the problem is related to becoming a Thread router as pointed out by Boris. Overall, the root cause of this issue seems to be that the device does not receive information about the SRP server as the following message does not show up: `SRP Client was started`. Hence the device does not register its services and the commissioner fails to discover it.

smishra-volansys, 2023-01-27:
Hey Damian,

1) The OTBR has been set up on a device using the imx6ull acting as a gateway. 
2) This happens every time we try to commission over ble-thread. (No successful commissioning yet)
3) Our setup is
-> Linux PC running the Chip-Tool
-> imx6ULL based gateway as OTBR
-> nrf52840 kit used as Matter Device (Lighting App)

4) I tried making the change as mentioned in point 4 still had no success. Please find the logs attached. 

[ble-thread-nrf-latest.log](https://github.com/project-chip/connectedhomeip/files/10519572/ble-thread-nrf-latest.log)
[ble-thread-chip-tool-latest.log](https://github.com/project-chip/connectedhomeip/files/10519576/ble-thread-chip-tool-latest.log)

Damian-Nordic, 2023-01-27:
This time it looks like the device is not even able to attach to the Thread network. Are you sure you used correct dataset? Anyway, I'm not familiar with OTBR on imx6ULL, but are you able to verify that OTBR on that platform enables SRP server and Advertising Proxy by default?

closed 2023-08-29 by smishra-volansys