project-chip/connectedhomeip #24701: [Tooling] No timestamps in cross-compiled chip-tool for arm64

lucicop, 2023-01-27:
### Reproduction steps / Feature

When cross-compiling chip-tool for arm64 from x64 host with command:
`./scripts/build/build_examples.py --target linux-arm64-chip-tool-ipv6only-clang build`
the wrong logging implementation is selected and there are no logging timestamps.

### Platform

raspi

### Type

CI tested

### (Optional) If manually tested please explain why this is only manually tested

Binary is produced with:
./scripts/build/build_examples.py --target linux-arm64-chip-tool-ipv6only-clang build

bzbarsky-apple, 2023-01-27:
This is presumably picking up the default stdio logging impl, which does not have cross-platform-enough ways to get timestamps.

bzbarsky-apple, 2023-05-25:
@ksperling-apple may remember how this stuff is set up.  But from a brief look, `src/platform/logging/BUILD.gn ` has:

```
static_library("stdio") {
  if (chip_device_platform == "darwin") {
    sources = [ "impl/stdio/darwin/Logging.cpp" ]
  } else {
    sources = [ "impl/stdio/Logging.cpp" ]
  }
```

and `examples/chip-tool/BUILD.gn` has:

```
  deps = [
    ":chip-tool-utils",
    "${chip_root}/src/platform/logging:force_stdio",
  ]
```

raising the question of how chip-tool on linux _ever_ gets timestamps, when not cross-compiled.

andy31415, 2024-06-20:
> @bzbarsky-apple would you have pointers on fixing this we can look at ? We would like to use chip-tool in a latency check, the timestamps are vital there.

You should probably use tracing infrastructure for latency checks. Logs are convenient because they exist, however tracing is the bit that we built to have machine-readable timestamps.

PR #34070 (2024-06-27): Unify stdio logging
Darwin now uses the same stdio logging implementation as other platforms.
Logs are colorized and contain timestamps.
On linux and darwin, pid and tid are also logged.

Fixes #24701
closed 2024-06-27 by mergify[bot]