project-chip/connectedhomeip #24711: [Feature] Darwin: Pass event timestamp to MTRDeviceDelegate

kpark-apple, 2023-01-27:
### Feature description

Current Darwin MTRDevice doesn't pass event timestamp to the MTRDeviceDelegate.
Apps may need to use the timestamp to discard too old events.
Please add event timestamps (maybe add to the dictionary with a dedicated key).

### Platform

darwin

bzbarsky-apple, 2023-01-27:
For an event, we should be including everything present in MTREventReport.  So:

1. MTREventPath
2. Event number
3. Event priority
4. Event timestamp
5. The actual payload

Or just the path and error if here is an error.

bzbarsky-apple, 2023-01-27:
Oh, and the event timestamp even in MTREventReport is kinda broken, since it just gives the value but not the type (system vs epoch time)....  We should add the type information in MTREventReport, and in the MTRDevice bits.

bzbarsky-apple, 2023-01-27:
And we should decide for "epoch time" whether to leave it as CHIP epoch time or convert it to something people are more likely to understand....

bzbarsky-apple, 2023-01-28:
Summary of Slack discussion, from @jtung-apple:
```
MTREventTimeTypeKey => NSNumber-wrapped enum of { epoch_type, system_type }
MTREventEpochTimeKey => NSDate object
MTREventSystemTimeKey => NSNumber-wrapped NSTimeInterval (which is seconds in a double)
```

and for MTREventReport, nullable NSDate and nullable NSNumber for the system time case.  And the enum.

PR #24858 (2023-02-10): Darwin: Pass event timestamp to MTRDeviceDelegate
This change adds event number, event priority, and event time to event reports with MTRDevice.

Fixes #24711
closed 2023-02-10 by jtung-apple