project-chip/connectedhomeip #24718: [BUG] Build Python fails due to Zap Generation on Raspberry Pi (no ARM based version of zap-cli available)

oidebrett, 2023-01-28:
### Reproduction steps

- Clone latest master branch of repo:
- git submodule update --init
- run scripts/build_python.sh -m platform -i separate 

Result: FAILED TO EXECUTE ZAP GENERATION

Error suggests:
* You may need to install zap. Please ensure one of these applies:
* - `zap-cli` is in $PATH. Install from https://github.com/project-chip/zap/releases

There is no compatible version of /zap/releases for ARM based architecture (i.e. Rasp Pi 4B+)

Can ZAP generation be skipped as we only require the python libraries so I can use chip-repl on the Rasp Pi? we would rather not have to download the source and compile the ZAP library.

### Bug prevalence

When building for Python platform on Raspberry Pi

### GitHub hash of the SDK that was being used

master

### Platform

raspi

### Platform Version(s)

master

bzbarsky-apple, 2023-01-28:
@oidebrett You should be able to do the pre-generation thing docs/code_generation.md describes, if the error is clear enough about which exact codegen bits you need.

> Can ZAP generation be skipped as we only require the python libraries

Those libraries depend on some generated things, right now.

Also note https://github.com/project-chip/zap/issues/897 but that does not help you for the moment.

bzbarsky-apple, 2023-01-28:
And see also https://github.com/project-chip/connectedhomeip/issues/24588#issuecomment-1400550663

zhouchunli, 2023-01-31:
same problem

oidebrett, 2023-01-31:
I've created a PR to provide a fix for this. Allowing pregenerated files in python build #24752. Hopefully I followed the correct procedures for creating a PR.

oidebrett, 2023-01-31:
Just to add that I tested the modified build_python.sh script as follows:

I pre-generated the files on a Ubuntu system that has zap-cli using
scripts/codepregen.py ${OUTPUT_DIRECTORY:-./zzz_pregenerated/}

I then transferred the zzz_pregenerated folder to the raspberry pi and ran
 scripts/build_python.sh -m platform -i separate -z ./zzz_pregenerated/

That worked and built the python environment without the need for the zap-cli tool installed on the raspberry pi

oidebrett, 2023-02-02:
This can be resolved by [building for arm64 eg. for raspberry pi](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/matter-repl.md#building-for-arm64-eg-for-raspberry-pi)

A new argument has been added to the build_python.sh to allow the use of pre-generated files.

e.g. scripts/build_python.sh -m platform -i separate -z ./zzz_pregenerated/

closed 2023-02-02 by oidebrett