project-chip/connectedhomeip #24734: Unable to compile all-clusters-app on M1 Mac

phonnakasturi-apple, 2023-01-30:
### Reproduction steps / Feature

With SHA 7237d349526575842a3d111d76aca759abe8d5f8

I am running into "CMake Error" when I execute idf.py

**Steps to reproduce:**

1. cd ~/connectedhomeip
2. git reset --hard 7237d349526575842a3d111d76aca759abe8d5f8
3. git clean -Xdf
4. git submodule update --init
5. export PKG_CONFIG_PATH="/opt/homebrew/opt/openssl@3/lib/pkgconfig"
6. source scripts/bootstrap.sh
7. source scripts/activate.sh
8. cd ~/tools/esp-idf
9. git pull
10. git checkout v4.4.3
11. git submodule update --init
12. ./install.sh
13. . ./export.sh
14. cd ~/connectedhomeip/examples/all-clusters-app/esp32
15. idf.py build

**Step 15 throws the following error:**

```
CMake Error at /Users/priyankahonnakasturi/tools/esp-idf/tools/cmake/build.cmake:452 (message):
Call Stack (most recent call first):
  /Users/priyankahonnakasturi/tools/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:38 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

```

**Note:** This error has been seen from the following SHA

commit 955f7f5881e575855447917a991d65c995cb4d1b
Author: Shubham Patil <shubham.patil@espressif.com>
Date:  Wed Jan 25 20:41:15 2023 +0530

**Attaching the complete log**
[cmake_error.txt](https://github.com/project-chip/connectedhomeip/files/10538227/cmake_error.txt)

### Platform

darwin

### Type

Platform validated

bzbarsky-apple, 2023-01-30:
@shubhamdp @dhrishi

shubhamdp, 2023-01-31:
@phonnakasturi-apple can you please share the version of idf-component-manager
```
python3 -m pip show idf-component-manager | grep Version
```

I am guessing its old, so may be try upgrading it
```
python3 -m pip install idf-component-manager --upgrade
```

Also, the script sourcing sequence should be idf first and connectedhomeip later. Since you have the environment setup,
```
# open a new terminal
cd esp-idf
. ./export.sh
cd connectedhomeip
source scripts/activate.sh
```

bzbarsky-apple, 2023-01-31:
> can you please share the version of idf-component-manager

Shouldn't `install.sh` handle that?

bzbarsky-apple, 2023-01-31:
So I tried this:
```
% cd examples/all-clusters-app/esp32
% source ~/tools/esp-idf/export.sh
% source ../../../scripts/activate.sh 
% % idf.py build
The following Python requirements are not satisfied:
idf-component-manager~=1.0
gdbgui==0.13.2.0
pygdbmi<=0.9.0.2
python-socketio<5
itsdangerous<2.1
kconfiglib==13.7.1
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
construct==2.10.54
To install the missing packages, please run "/Users/bzbarsky/tools/esp-idf/install.sh"

But I absolutely ran `install.sh` before all this (though not in the "activate" environment, because `docs/guides/esp32/setup_idf_chip.md` certainly does not mention activate anywhere in that part).

What are the actual steps for setting up an ESP32 build environment and building, and can we please have the documentation clearly list those steps?

phonnakasturi-apple, 2023-01-31:
@shubhamdp idk version is 
```
python3 -m pip show idf-component-manager | grep Version
Version: 1.0.1
```

dhrishi, 2023-02-01:
Thanks for the update @phonnakasturi-apple. We will close this after we verify and update the documentation flow. cc @shubhamdp

shubhamdp, 2023-02-02:
Performed following steps on Intel chip macbook
```
cd /tmp
git clone -b v4.4.3 --recursive --depth 1 --shallow-submodules https://github.com/espressif/esp-idf.git
git clone --recursive --depth=1 --shallow-submodules https://github.com/project-chip/connectedhomeip.git

cd esp-idf && ./install.sh && cd -
cd connectedhomeip && source scripts/bootstrap.sh && cd -

source esp-idf/export.sh
source connectedhomeip/scripts/activate.sh

cd connectedhomeip/examples/all-clusters-app/esp32
idf.py build
```

I get the same error as mentioned by Boris in https://github.com/project-chip/connectedhomeip/issues/24734#issuecomment-1410497666.

This is because https://github.com/project-chip/connectedhomeip/pull/22071 restricted `requirements.esp32.txt` to only linux platform. (One more ref comment from @andy31415  https://github.com/project-chip/connectedhomeip/issues/23477#issuecomment-1313978928).

Using `scripts/build/build_examples.py --target esp32-m5stack-all-clusters build` also fails with the same errors.

Problem is with the `gevent`.

closed 2023-02-11 by andy31415