project-chip/connectedhomeip #24735: Consider some sort of setup where we browse for operational advertisements and act on them

bzbarsky-apple, 2023-01-30:
We could browse for `_matter._tcp` and if a new instance name appears possibly act on it by, for example:
1. Retriggering any pending resubscribe attempts to the given node id.
2. Notifying higher layers that the node id is now visible.
3. What else?

Not clear, if we do this, whether this should live in the core SDK, or the Darwin framework, or somewhere else.

In practical terms, right now we do not have a way to "browse without doing a resolve", nor do we have a way to "stop a single resolve for the given browse", so we would need to think about the resolve lifetimes here a bit.

@jtung-apple @woody-apple @vivien-apple

ndyck14, 2023-01-30:
a more minor optimization, but for cases like commissioning, this would speed up the process as callbacks can trigger the resolver instead of a timer.

jonsmirl, 2023-02-10:
After a device reboots, I can use the Home app to tap on the device sending it commands, and then watch the device discard them because they were sent on the old session,  So it would be useful if the the Homepod could use MDNS to recognize the device reboot and establish a new session.

bzbarsky-apple, 2023-02-10:
I don't think  you can really identify "device reboot" via MDNS....  At least not that I can tell.

ndyck14, 2023-02-10:
it is not necessary to understand the reason for a DNS announcement outside of the fact it (should) imply the device is present. for the sake of a CASE session that has lapsed (e.g. due to maxInterval on subscription), it would be wise to attempt to re-establish a new CASE session (at least until the proper response to a stale session is implemented)

bzbarsky-apple, 2023-02-10:
> it is not necessary to understand the reason for a DNS announcement outside of the fact it (should) imply the device is present.

Yes....

> for the sake of a CASE session that has lapsed (e.g. due to maxInterval on subscription)

This does not "lapse" a CASE session, though a timeout when trying to use one does.  But this does not handle the "reboot" case @jonsmirl is talking about either.

bzbarsky-apple, 2023-02-10:
I know what problem you are trying to fix.  I'm saying that mdns does not let you fix that problem.

jonsmirl, 2023-02-10:
The problem going to occur when a customer is having setup issues and they keep rebooting the device trying to fix it. By rebooting the device they will keep retriggering the session time outs. Worst case they end up deciding the device is broken and return it.

jonsmirl, 2023-02-10:
At boot, could the device look an see if there is an existing MDNS entry? And then increment a version count in it if it finds one?

jonsmirl, 2023-02-10:
Maybe this TSR record could be used to detect reboot? It is Apple RFC
http://www.watersprings.org/pub/id/draft-tllq-tsr-02.html

bzbarsky-apple, 2023-02-11:
Can we please put the proposed spec change discussions in spec issues, not here?

ndyck14, 2023-02-15:
> This does not "lapse" a CASE session, though a timeout when trying to use one does. But this does not handle the "reboot" case @jonsmirl is talking about either.
In practicality, as soon as a device crashes or powers off, the CASE session is lapsed even if a client does not know. DNS advs will not solve a reboot case entirely, although using it as a "hint" to check if a CASE session is still live is valuable. For example, this could trigger a read if the session is presumed still active. If a single read times out, it makes sense to attempt to re-establish the session. If the client side session has already lapsed due to timeout on subscription or other client specific logic, then it makes sense to immediately attempt to establish a new CASE session

this seems like a significant improvement on the current detection mechanism which seems to basically be a perpetual polling (depending on implementation).

jonsmirl, 2023-02-15:
One idea, which needs a spec change, is for the device to make a subDNS tag encoding its boot time when it boots. If this tag is greater than what you saved, a reboot has happened. Use 32b hex for seconds since 2022 or similar,  like T00CB45AB. This also has the side effect of providing a way to compute uptime for a device.

bzbarsky-apple, 2023-02-15:
Closing issue, since all the spec-change noise is making it unusable.  If  you have spec changes to propose, please do not propose them in SDK issues, where they are not actionable.

closed 2023-02-15 by bzbarsky-apple