project-chip/connectedhomeip #24740: Cross compiling for RPI causes errors

tobyloki, 2023-01-30:
Hello. I am trying to cross compile the chip-tool on my Ubuntu 22.04.1 desktop for the Raspberry PI which is running on aarch64. Here are the steps I've completed so far.

Install some dependencies
```cmd
sudo apt install gcc-aarch64-linux-gnu
sudo apt install g++-aarch64-linux-gnu
```

Create symbolic links for openssl files that the build command (next step) complained about missing openssl/opensslconfig.h
```cmd
cd /usr/include/openssl/
sudo ln -s /usr/include/gnutls/openssl.h .
sudo ln -s ../x86_64-linux-gnu/openssl/opensslconf.h .
sudo ln -s ../x86_64-linux-gnu/openssl/configuration.h .
```

Build chip-tool
```cmd
./scripts/examples/gn_build_example.sh examples/chip-tool out/rpi 'target_os="linux" target_cpu="arm64"'
```

Towards the end of build, I get the following errors. How can I fix them?
```
[1/2] ld ./chip-tool
FAILED: chip-tool chip-tool.map 
aarch64-linux-gnu-g++ -O0 -fPIC -Werror -Wl,--fatal-warnings -Wl,-z,defs -fdiagnostics-color -Wl,--gc-sections -pie liblogging-stdio.a -Wl,-Map,./chip-tool.map @./chip-tool.rsp -o ./chip-tool
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: cannot find -lssl: No such file or directory
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: cannot find -lcrypto: No such file or directory
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: cannot find -lgio-2.0: No such file or directory
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: cannot find -lgobject-2.0: No such file or directory
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: cannot find -lglib-2.0: No such file or directory
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

bzbarsky-apple, 2023-01-30:
> Create symbolic links for openssl files that the build command (next step) complained about missing 

You need aarch64 versions of those libraries, no?  Symlinking the x86_64 versions of the headers might work for the compile (though might also do weird things), but the link step is failing because the actual libraries are not present.

tobyloki, 2023-01-30:
> but the link step is failing because the actual libraries are not present.

How do I get the actual libraries on my Ubuntu machine?

bzbarsky-apple, 2023-01-30:
Probably from Ubuntu packages for arm64 (e.g. https://packages.ubuntu.com/bionic/arm64/libssl-dev/filelist) that you would need to install non-globally somewhere...

rosahay-silabs, 2023-02-06:
Duplicate of #21853

The steps mentioned there work for WSL2 and logically should work for any Linux installation.

bzbarsky-apple, 2023-02-06:
Duplicate of #21853

closed 2023-02-06 by bzbarsky-apple