project-chip/connectedhomeip #24778: [TC-ACE-1.2] SDK suport required to send a subscribtion when two controllers are in same fabric.

Rajashreekalmane, 2023-02-01:
**Test Case ID :-** 
4.1.2. [TC-ACE-1.2] Subscriptions

**Here explaining two scenario's** 

1. **scenario 1**

As per Test-plan, If TH1 and TH2 are in same fabric and tried to subscribe an attribute, DUT through an error as Status = 0x80 (INVALID_ACTION) instead of 0x00 (SUCCESS)

**Expected  behavior** :- 0x00 (SUCCESS)
**Actual behavior** :-  0x80 (INVALID_ACTION)

**Reproduction steps : 
On DUT** 
./chip-all-clusters-app
**On TH1** 
./chip-tool pairing onnetwork 1 20202021 --commissioner-nodeid 0x1 
 ./chip-tool interactive start   
accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": [{ "cluster": null, "endpoint": 0, "deviceType": null }]}]' 1 0   --commissioner-nodeid 0x1

**On TH2(same fabric as N1)** 
operationalcredentials subscribe nocs  1 30 1 0 --fabric-filtered false  --keepSubscriptions 0    --commissioner-nodeid 0x2

**Test-Plan Reference :** 
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/AccessControlEnforcement.adoc#tc-ace-1-2-subscriptions-removed

**PFA log below** 
[TC-ACE-1.2-DUT.txt](https://github.com/project-chip/connectedhomeip/files/10553833/TC-ACE-1.2-DUT.txt)
[TC-ACE-1.2-TH.txt](https://github.com/project-chip/connectedhomeip/files/10553834/TC-ACE-1.2-TH.txt)

2. **scenario 2** 

As per Test-plan, If TH1 and TH2 are in same fabric and tried to send ACL write cammand, DUT through an error as Status = 0x80 0x7e (UNSUPPORTED_ACCESS ) instead of 0x00 (SUCCESS)

**Reproduction steps : 
On DUT** 
./chip-all-clusters-app
**On TH1** 
./chip-tool pairing onnetwork 1 20202021 
 ./chip-tool interactive start   
accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": [{ "cluster": null, "endpoint": 0, "deviceType": null }]}]' 1 0   --commissioner-nodeid 0x1

**Expected  behavior** :- 0x00 (SUCCESS)
**Actual behavior** :- 0x7e (UNSUPPORTED_ACCESS )

**PFA log below** 
[TC-ACE-1.2----DUT.txt](https://github.com/project-chip/connectedhomeip/files/10553853/TC-ACE-1.2----DUT.txt)
[TC-ACE-1.2---TH.txt](https://github.com/project-chip/connectedhomeip/files/10553855/TC-ACE-1.2---TH.txt)

lisaaaas, 2023-02-01:
I also encountered the same problem. When I executed instruction “operationalcredentials subscribe nocs  1 30 2 0 --fabric-filtered false  --keepSubscriptions 0”, the result was timeout, and I found that my instruction was different from yours, yours was accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": [{ "cluster": null, "endpoint": 0, "deviceType": null }]}]' 1 0 --commissioner-nodeid 0x1. Is this the instruction in the latest test document?
[image]

bzbarsky-apple, 2023-02-01:
>  `accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": [{ "cluster": null, "endpoint": 0, "deviceType": null }]}]' 1 0 --commissioner-nodeid 0x1`

This does not match the test plan.  Test plan says:

> Subjects field: [N1, N2]

but you are including only N1.  Then when N2 tries to subscribe, it gets UNSUPPORTED_ACCESS, because it does not have access.

bzbarsky-apple, 2023-02-01:
I filed https://github.com/CHIP-Specifications/chip-test-plans/issues/2417 on the verification steps being wrong.

closed 2023-02-01 by bzbarsky-apple

lisaaaas, 2023-02-02:
> Ok, so is the test case TC-ACE-1.2 currently problematic or has it been updated with the latest test documentation?