project-chip/connectedhomeip #24783: [BUG] dynamic-bridge-app: Aggregator PartList attribute empty

raven-worx, 2023-02-01:
### Reproduction steps

Inspecting my custom bridge (based on dynamic-bridge-app) i noticed that the Aggregator (EP 1) Endpoint's PartList attribute is empty, instead of being filled with the available bridged device node endpoints. Only the Root Nodes PartList attribute is filled.

According to the Core Spec (9.12.2):
> Additionally, it SHALL expose an endpoint with the device type Aggregator which has a Descriptor
cluster with a PartsList attribute containing all the endpoints representing those Bridged
Devices. See Endpoint Composition for the concept of hierarchical composition.

It seems Google Home doesn't mind this and displays my endpoints, whereas Apple Home doesn't like it and displays no endpoints at all.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

43dd327f1cf497cdf8c90dfb2e481dd567abfaa9

### Platform

core

### Platform Version(s)

1.0.0.2

bzbarsky-apple, 2023-02-01:
The "special handling" needs to be in the example dynamic-bridge-app.  It should be setting up its endpoints with the aggregator as the parent, and it's not.

In particular, in `AddDeviceEndpoint` it does:
```
                ret =
                    emberAfSetDynamicEndpoint(index, gCurrentEndpointId, dev->endpointType(), dev->versions(), dev->deviceTypes());
```

where the function is declared as:
```
EmberAfStatus emberAfSetDynamicEndpoint(uint16_t index, chip::EndpointId id, const EmberAfEndpointType * ep,
                                        const chip::Span<chip::DataVersion> & dataVersionStorage,
                                        chip::Span<const EmberAfDeviceType> deviceTypeList = {},
                                        chip::EndpointId parentEndpointId                  = chip::kInvalidEndpointId);
```

Note the lack of `parentEndpointId` at the callsite.

bzbarsky-apple, 2023-02-01:
@andy31415 could you please assign to @dnicoara?

bzbarsky-apple, 2023-08-07:
dynamic-bridge is gone.

closed 2023-08-07 by bzbarsky-apple