project-chip/connectedhomeip #24840: [chiptest] ble-wifi Pairing failed while using test harness in Raspberrypi 4b.

fufenxiang, 2023-02-03:
Problem
I built the test-harness env in raspberrypi 4b, run tc-oo-1.1 case, the th web report 'utf-8' codec can't decode byte error, commision fails.

Test Project config as follow:
{
  "config": {
    "network": {
      "wifi": {
        "ssid": "matter_6600",
        "password": "12345678"
      },
      "thread": {
        "rcp_serial_path": "/dev/ttyACM0",
        "rcp_baudrate": 115200,
        "on_mesh_prefix": "fd11:22::/64",
        "network_interface": "wlan0",
        "dataset": {
          "channel": "15",
          "panid": "0x1234",
          "extpanid": "1111111122222222",
          "networkkey": "00112233445566778899aabbccddeeff",
          "networkname": "DEMO"
        },
        "otbr_docker_image": null
      }
    },
    "dut_config": {
      "discriminator": "248",
      "setup_code": "7636530",
      "pairing_mode": "ble-wifi",
      "chip_tool_timeout": null
    },
    "test_parameters": null
  },
  "pics": {
    "clusters": {}
  }
}

the logs:
[UI_Test_Run_G_Commissioner_2023_2_2_10_25_12.log](https://github.com/project-chip/connectedhomeip/files/10577040/UI_Test_Run_G_Commissioner_2023_2_2_10_25_12.log)

Do i need to modify raspi's codec configuration?

bzbarsky-apple, 2023-02-03:
@fufenxiang The right place for testharness issues is https://github.com/CHIP-Specifications/chip-certification-tool

fufenxiang, 2023-02-06:
> @fufenxiang The right place for testharness issues is https://github.com/CHIP-Specifications/chip-certification-tool
ok, thank you.

fufenxiang, 2023-02-06:
@bzbarsky-apple ok, thank you.

cjandhyala, 2023-03-06:
not an sdk issue, closing it here. pls open ticket in TH repo.

closed 2023-03-06 by cjandhyala