project-chip/connectedhomeip #24841: [BUG] Conflicting dependency error

mustafayuce33, 2023-02-03:
### Reproduction steps

I tried to build dynamic-bridge-app but I get dependency error. Here is the error that I get;

```
Setting up CIPD package manager...done (17.7s)
  Setting up Python environment.....Ignoring pyobjc-core: markers 'sys_platform == "darwin"' don't match your environment
Ignoring pyobjc-framework-cocoa: markers 'sys_platform == "darwin"' don't match your environment
Ignoring pyobjc-framework-corebluetooth: markers 'sys_platform == "darwin"' don't match your environment
Collecting pip-tools
  Using cached pip_tools-6.1.0-py3-none-any.whl (44 kB)
Collecting virtualenv
  Using cached virtualenv-20.4.7-py2.py3-none-any.whl (7.2 MB)
ERROR: Cannot install click>=7.0 because these package versions have conflicting dependencies.

The conflict is caused by:
    The user requested click>=7.0
    The user requested (constraint) click==8.1.3

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/user_guide/#fixing-conflicting-dependencies
['/data/jenkins/jobs/thinka-matter/workspace/src/dynamic-bridge/third_party/connectedhomeip/.environment/pigweed-venv/bin/python', '-m', 'pip', 'install', '--log', '/data/jenkins/jobs/thinka-matter/workspace/src/dynamic-bridge/third_party/connectedhomeip/.environment/pigweed-venv/pip-requirements.log', '--requirement=/data/jenkins/jobs/thinka-matter/workspace/src/dynamic-bridge/third_party/connectedhomeip/scripts/requirements.txt', '--constraint=/data/jenkins/jobs/thinka-matter/workspace/src/dynamic-bridge/third_party/connectedhomeip/scripts/constraints.txt'] {'stdout': <_io.TextIOWrapper name=3 mode='w+' encoding='UTF-8'>, 'stderr': -2}
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/tree/2ce5f2d6e88a69e24ca8542cd03c233292755423

### Platform

raspi

mustafayuce33, 2023-02-03:
Can you fix here https://github.com/project-chip/connectedhomeip/blob/master/scripts/constraints.txt#L40. It says `click` should be equal to `8.1.3`. But on the other hand here https://github.com/project-chip/connectedhomeip/blob/master/scripts/requirements.esp32.txt#L1 it says it should be larger than `>=7.0`

mustafayuce33, 2023-02-03:
And also it gives the same error for some other requirements as well. This is another one;

```
ERROR: Cannot install ipython because these package versions have conflicting dependencies.

The conflict is caused by:
    The user requested ipython
    The user requested (constraint) ipython==7.24.1
```

bzbarsky-apple, 2023-02-03:
@shubhamdp @dhrishi @andy31415

andy31415, 2023-02-03:
By the looks of the log, this seems to happen within `bootstrap.sh`. Is that right?

@mustafayuce33 please provide more info to see how one reproduces this:

- what exact platform? Raspi is the hadrware .... what ubuntu version did you use, what were the steps to setup this machine
- confirm if this happens just by executing "boostrap"
- how did you setup things? i.e. is this a fresh checkout?
- what is your (if any) $PW_ENVIRONMENT_ROOT? if non-empty maybe try to clear it out first
- The logs say `/data/jenkins/jobs/` ... which seems like some form of automation. Is this reproducible on a fresh machine? I am wondering if we would need to narrow down jenkins vs fresh RPi machine differences.

andy31415, 2023-02-03:
I also find the paths mentioned: `/data/jenkins/jobs/project/workspace/src/dynamic-bridge/third_party/connectedhomeip/.environment/pigweed-venv/bin/python` suspect ... it goes through a third-party indirection, which I find odd, but maybe it is expected, unsure.

What command are you using to try the build? Please use something of the form `./scripts/build/build_examples.py --target linux-arm64-dynamic-bridge build`

andy31415, 2023-02-03:
For me, on a x64 (so cross compile) these worked fine on the vscode chip image:

```
./scripts/build/build_examples.py --target linux-arm64-dynamic-bridge-clang build
./scripts/build/build_examples.py --target linux-x64-dynamic-bridge build
```

mustafayuce33, 2023-02-03:
@andy31415 Here is the information about my build machine; 

```
NAME="Ubuntu"
VERSION="18.04.6 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.6 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

mustafayuce33, 2023-02-03:
And I try to build for Raspi on this Ubuntu machine. I downloaded all the files on Ubuntu machine (under `/opt/pi/sysroot`) with rsync command. And then I execute the following commands

```
$ source ./scripts/bootstrap.sh (or active.sh. I guess those are the same)
$ gn_build_example.sh examples/dynamic-bridge-app out/rpi "target_cpu="arm" target_os="linux" sysroot="$HOME/pi/sysroot" system_libdir="lib/arm-linux-gnueabihf" chip_config_network_layer_ble=false is_clang=true"
```
The problem is bootstrap gives this error.

andy31415, 2023-02-03:
Please try with:

```
./scripts/build/build_examples.py --target linux-arm64-dynamic-bridge-clang build
```

I am unsure about gn_build_example and those are a lot of arguments to pass in to debug.

andy31415, 2023-02-03:
actually if boostrap gives the error, that is something else  ... it does not even get to the actual build.

mustafayuce33, 2023-02-03:
It give an error 

```
/usr/bin/env: invalid option -- 'S'
Try '/usr/bin/env --help' for more information.
```

mustafayuce33, 2023-02-03:
> actually if boostrap gives the error, that is something else ... it does not even get to the actual build.
Exactly

mustafayuce33, 2023-02-03:
I think if you run `bootstrap.sh` on `Ubuntu 18.04.6 LTS` you will see the same result

andy31415, 2023-02-03:
Your ubuntu seems old: 18.

Our docs say 22.04: https://project-chip.github.io/connectedhomeip-doc/guides/BUILDING.html

I think 20.04 also works because CI is https://github.com/project-chip/connectedhomeip/blob/master/integrations/docker/images/chip-build/Dockerfile#L2

However 18 is too old.

mustafayuce33, 2023-02-03:
It is a pitty because this machine is our CI/CD so we can not upgrade it easily

mustafayuce33, 2023-02-03:
BTW I can build it on both Ubuntu `20.4` and `22.04`

andy31415, 2023-02-03:
What this means is that debugging may be up to you - we will accept patches for more compatibility (assuming they do not break existing things), however your environment is likely different from what others (including github CI) uses.

mustafayuce33, 2023-02-03:
I guess it is more related with python version.

andy31415, 2023-02-03:
Since the error is saying click 8 does not satisfy (>= 7) that sounds like an oddity. You may be able to update your packages somehow - if you have a system click, uninstall it or maybe upgrade pip. All would be trial and error though.

Another idea would be to install a separate python (from source?) somewhere else and make the build use that one.

mustafayuce33, 2023-02-03:
I built python 3.10 for that machine but could find a way for connectedhomeip to use it. do you have any idea?

mustafayuce33, 2023-02-03:
I guess python 3.6 gives an error if requirement says click==8 and the other one says (>= 7). But for 3.10 it is ok

mustafayuce33, 2023-02-03:
So I need to find someway to force connectedhomeip to use my python executable instead of itself

andy31415, 2023-02-03:
Create a virtualenv and before invoking bootstrap/activate/build source it's activate.sh

mustafayuce33, 2023-02-03:
Can you make it much more clear about how to do it?

andy31415, 2023-02-03:
So I was thinking to use https://docs.python.org/3/library/venv.html

Basically:

- install python3 to a newer version somewhere, maybe from source
- run `$INSTALL_PATH/bin/python3 -m venv /opt/chip-python-venv`
- then when you need to compile:

```
source /opt/chip-python-venv/bin/activate
source scripts/bootstrap.sh
./scripts/build/build_examples.py --target linux-arm64-dynamic-bridge-clang build
```

and I hope the above has a chance to work.
Basically source the `activate` script every time before invoking chip builds, so the right python environment is used.
That or set the new python to the front of `$PATH`

mustafayuce33, 2023-02-03:
Thanks I will try these

mustafayuce33, 2023-02-03:
All requirement errors have been resolved but it could not build a library

mustafayuce33, 2023-02-03:
Is there any docker file for building?

andy31415, 2023-02-03:
chip-build-crosscompile image will be able to cross compile.

With gevent, the dependency is esp32 needs gdb-gui which wants gevent. Since you do not care about esp32, you could just patch `scripts/requirements.esp32.txt` and comment out `gdb-gui`. Hackish, but it is something I also had to do on my arch system :(

mustafayuce33, 2023-02-07:
@andy31415 Finally I've built the project. Thanks for your great help.

closed 2023-02-10 by mustafayuce33