project-chip/connectedhomeip #24855: [BUG] Failed Ninja compile step with arm64-apple-12.0 Apple Silicon M1 chip

Quynh-Nguyen, 2023-02-03:
### Reproduction steps

1. Run the command
```
ninja -C out/custom
```

2. Got the error
```
ninja: Entering directory `out/custom'
[1/1350] c++ obj/src/app/util/all-clusters-common.error-mapping.cpp.o
FAILED: obj/src/app/util/all-clusters-common.error-mapping.cpp.o
g++ -MMD -MF obj/src/app/util/all-clusters-common.error-mapping.cpp.o.d -Wconversion -target arm64-apple-12.0 -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -I/opt/homebrew/Cellar/openssl@3/3.0.7/include -fobjc-arc -std=gnu++14 -frtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\<lib/address_resolve/AddressResolve_DefaultImpl.h\> -I../../zzz_generated/all-clusters-app -Igen/examples/all-clusters-app/all-clusters-common -Igen/examples/all-clusters-app/all-clusters-common/zapgen -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -c ../../src/app/util/error-mapping.cpp -o obj/src/app/util/all-clusters-common.error-mapping.cpp.o
In file included from ../../src/app/util/error-mapping.cpp:18:
In file included from ../../src/app/util/error-mapping.h:21:
../../src/protocols/interaction_model/Constants.h:29:10: fatal error: 'type_traits' file not found
#include <type_traits>
         ^~~~~~~~~~~~~
1 error generated.
[2/1350] c++ obj/src/app/util/all-clusters-common.attribute-size-util.cpp.o
FAILED: obj/src/app/util/all-clusters-common.attribute-size-util.cpp.o
g++ -MMD -MF obj/src/app/util/all-clusters-common.attribute-size-util.cpp.o.d -Wconversion -target arm64-apple-12.0 -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -I/opt/homebrew/Cellar/openssl@3/3.0.7/include -fobjc-arc -std=gnu++14 -frtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\<lib/address_resolve/AddressResolve_DefaultImpl.h\> -I../../zzz_generated/all-clusters-app -Igen/examples/all-clusters-app/all-clusters-common -Igen/examples/all-clusters-app/all-clusters-common/zapgen -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -c ../../src/app/util/attribute-size-util.cpp -o obj/src/app/util/all-clusters-common.attribute-size-util.cpp.o
In file included from ../../src/app/util/attribute-size-util.cpp:18:
In file included from ../../src/app/util/af.h:36:
In file included from ../../src/app/util/af-types.h:31:
In file included from ../../src/app/util/basic-types.h:30:
../../src/lib/core/DataModelTypes.h:20:10: fatal error: 'cstdint' file not found
#include <cstdint>
         ^~~~~~~~~
1 error generated.
[3/1350] c++ obj/src/app/util/all-clusters-common.ember-print.cpp.o
FAILED: obj/src/app/util/all-clusters-common.ember-print.cpp.o
g++ -MMD -MF obj/src/app/util/all-clusters-common.ember-print.cpp.o.d -Wconversion -target arm64-apple-12.0 -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -I/opt/homebrew/Cellar/openssl@3/3.0.7/include -fobjc-arc -std=gnu++14 -frtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\<lib/address_resolve/AddressResolve_DefaultImpl.h\> -I../../zzz_generated/all-clusters-app -Igen/examples/all-clusters-app/all-clusters-common -Igen/examples/all-clusters-app/all-clusters-common/zapgen -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -c ../../src/app/util/ember-print.cpp -o obj/src/app/util/all-clusters-common.ember-print.cpp.o
../../src/app/util/ember-print.cpp:18:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
```

### Bug prevalence

When ever I do it

### GitHub hash of the SDK that was being used

cd919db9a413dd768c76ecfcadd6ff66f691e5ca

### Platform

esp32

bzbarsky-apple, 2023-02-04:
@dhrishi @shubhamdp

dhrishi, 2023-02-07:
@Quynh-Nguyen  Which specific IDF v4.4 release or branch do you use? Currently, we support [IDF v4.4.3 release](https://github.com/espressif/esp-idf/releases/tag/v4.4.3) with Matter. Please use the same and share the steps you've followed along with the logs.

dhrishi, 2023-02-07:
Steps to be followed are mentioned here: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/README.md

Quynh-Nguyen, 2023-02-07:
@dhrishi 
Yes, I used the 4.4.3 version.
I found the issue that the python packages dependencies did not match with Python 3.9 version.
I downgraded to Python 3.8, and it's working well now.

bzbarsky-apple, 2023-02-08:
Wait, why is this closed?  Python 3.9 is the default on MacOS at this point, and needs to work....

dhrishi, 2023-02-08:
@bzbarsky-apple Okay. Makes sense

woody-apple, 2023-06-23:
@Quynh-Nguyen is this still happening for you?

woody-apple, 2023-06-24:
Please re-open if so.

closed 2023-06-24 by woody-apple

jvanmourik, 2023-07-18:
I have this problem on my M1 chip aswell. Using python 3.9.6 (default) and idf 5.1

shubhamdp, 2023-07-19:
@jvanmourik can you please check with the setup steps mentioned here: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/setup_idf_chip.md

jvanmourik, 2023-07-20:
> @jvanmourik can you please check with the setup steps mentioned here: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/setup_idf_chip.md

I could not get it to work with those directions unfortunately. Updating my python to version 3.11 did the trick however. Ideally this should work with python 3.9 which is the out-of-the-box version in MacOS Ventura