project-chip/connectedhomeip #24863: [Documentation] HVAC Device Types missing documentation

carlessole, 2023-02-05:
### Documentation issues

I have a technical question about Matter Specification.
I have taken a look at 'Matter Device Library" and I'm unable to understand why the "Type 9.1) Heating/Cooling Unit" implements the cluster Thermostat as Client.
Te¡hen, Matter Controller can not set the "Heating/Cooling Unit" to the desired Setpoint or Mode as the related commands direction are from "Client -> Server" as far as I have understood.
I'm not understanding anything maybe?
Could you explain? Where I can find more information about that?
Many thanks,
Carles

bzbarsky-apple, 2023-02-06:
"Heating/Cooling Unit" is sort of meant to be something like a space heater, or air conditioner, or furnace.  It has Thermostat as client so it can connect to a Thermostat server and discover things like what the set point and current temperature are and hence whether it should be operating.

But also, generally spec issues should be filed in https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/, not here....

closed 2023-02-06 by bzbarsky-apple

carlessole, 2023-02-06:
Thanks @bzbarsky-apple 
I have tried to open your link (https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/) but seems to be not workable.
Then I've found this (https://github.com/CHIP-Specifications/connectedhomeip) but I'm not able to see an "issues" option and https://github.com/CHIP-Specifications/connectedhomeip/issues is redirecting me to  https://github.com/CHIP-Specifications/connectedhomeip/pulls. So, I don't know where specs issues needs to be filled.

Then, regarding your answer, a "Heating/Cooling Unit" can not operate without a "Thermostat" device on the bus from where it reads the "setpoint" and the other signals?
I mean, a Matter network with just CHIP-Tool (or Google Home as main controller), there is no way to control the "Heating/Cooling Unit" setpoint without another "Thermostat" device in the bus?

Thanks again for the answer

bzbarsky-apple, 2023-02-06:
@carlessole The link I gave you is the right link, but might be restricted to Communications Standards Alliance members...

> Then, regarding your answer, a "Heating/Cooling Unit" can not operate without a "Thermostat" device on the bus from where it reads the "setpoint" and the other signals?

It can.  For example, some other entity could explicitly turn the heating/cooling unit on and off (and control its "Level Control" cluster, if that's implemented), instead of binding it to a thermostat.

> there is no way to control the "Heating/Cooling Unit" setpoint 

"Heating/Cooling Unit" does not have a concept of "setpoint" at all.  It has controllable on/off state, must be able to talk 
to a thermostat if bound to one, and may optionally support Level Control for more fine-grained control than just on/off.

carlessole, 2023-02-06:
Ok, thanks @bzbarsky-apple.

I'm used to develop many AC Interfaces using other standard protocols as KNX, BACnet, Modbus... where I can define the endpoints (not Matter endpoints, but the final communication objects, registers...) I desired. And it's hard for me to understand how to deal with (control & monitor) these typical AC Indoor Unit signals in Matter:

- OnOff (That one, ok => using the OnOff Cluster)
- UserMode (??)
- Setpoint (??)
- ReferenceTemp/RoomTemp (??)
- Fan Speed (??)
- Louvers (??)
- etc

bzbarsky-apple, 2023-02-06:
A "Heating/Cooling Unit" doesn't really represent an entire AC system.  It might represent just the "compressor and blower" part, perhaps.  So it has no concept of setpoint, user mode, reference temperature, or any of that.

There is work being done to define "Room Air Conditioner" devices that would have temperature sensing bits and whatnot.  But that work has not been finished yet.

carlessole, 2023-02-07:
Many thanks again.
I will be watching for the updates :)
I am still not understanding how a "HeatCool Unit" uses the Thermostat (as client) for its internal behavior. Is there any guide where I can find it?
Or currently there are just the official specifications ?

bzbarsky-apple, 2023-02-07:
@carlessole I don't know of any guides, sorry....

But one way it could work is this.  The Heat/Cool Unit reads the LocalTemperature attribute from the thermostat server.  The Heat/Cool Unit reads the OccupiedHeatingSetpoint from the thermostat server.  If LocalTemperature < OccupiedHeatingSetpoint, the unit turns on and heats, until LocalTemperature >= OccupiedHeatingSetpoint.

carlessole, 2023-02-07:
Great!
All your answers have been very useful!
We are eagerly looking forward to start developing Matter devices