project-chip/connectedhomeip #24864: [BUG] xcode14.2 build Fail

15202284857, 2023-02-06:
### Reproduction steps

[image]

### Bug prevalence

whenever i do this

### GitHub hash of the SDK that was being used

0

### Platform

darwin

bzbarsky-apple, 2023-02-06:
@15202284857 what actual commit are you using (certainly not 0) and what exact commands are you running?

Or if you are building from the Xcode UI, what exact thing are you building?

15202284857, 2023-02-06:
[image]
Based on the above flow,xcode will not run

bzbarsky-apple, 2023-02-06:
@15202284857 You did not answer my first question.  Again, which actual commit are you using?  Because if I `git clone` this morning, with commit `5c8ae45f7ab4f3bbb631eed8c43a51d81f42ca6b` into `/tmp/connectedhomeip`, then the checkout step succeeds fine...  And before we dive into debugging based on line numbers, I need to know what exact source code your are running.

bzbarsky-apple, 2023-02-06:
And just to check.  If you just run this in your checkout:
```
scripts/checkout_submodules.py --shallow --platform darwin
```
do you get the same error?