project-chip/connectedhomeip #24868: Error of compilation of lighting-app

esliu-apple, 2023-02-06:
### Reproduction steps

With Matter SHA fc41449659582354b44ed92905a6dc6ef9c9c227.

I ran into following error:

`[505/527] Building CXX object CMakeFiles/app.dir/Users/emmanuelliu/connectedhomeip/src/app/util/attribute-storage.cpp.obj
FAILED: CMakeFiles/app.dir/Users/emmanuelliu/connectedhomeip/src/app/util/attribute-storage.cpp.obj
ccache /Users/emmanuelliu/chip-nordic-tools/gnuarmemb/bin/arm-none-eabi-g++ -DCHIP_HAVE_CONFIG_H -DKERNEL -DMBEDTLS_CONFIG_FILE=\"nrf-config.h\" -DMBEDTLS_USER_CONFIG_FILE=\"nrf-config-user.h\" -DNRF52840_XXAA -DNRF_802154_ACK_TIMEOUT_ENABLED=1 -DNRF_802154_CCA_CORR_LIMIT_DEFAULT=2 -DNRF_802154_CCA_CORR_THRESHOLD_DEFAULT=45 -DNRF_802154_CCA_ED_THRESHOLD_DEFAULT=45 -DNRF_802154_CCA_MODE_DEFAULT=NRF_RADIO_CCA_MODE_ED -DNRF_802154_ECB_PRIORITY=-1 -DNRF_802154_ENCRYPTION_ENABLED=1 -DNRF_802154_IE_WRITER_ENABLED=1 -DNRF_802154_INTERNAL_RADIO_IRQ_HANDLING=0 -DNRF_802154_PENDING_EXTENDED_ADDRESSES=16 -DNRF_802154_PENDING_SHORT_ADDRESSES=16 -DNRF_802154_RX_BUFFERS=16 -DNRF_802154_SECURITY_WRITER_ENABLED=1 -DNRF_802154_SWI_PRIORITY=1 -DNRF_802154_TX_STARTED_NOTIFY_ENABLED=1 -DNRF_802154_USE_RAW_API=1 -DUSE_PARTITION_MANAGER=1 -DZCBOR_ASSERTS -D_ANSI_SOURCE -D_DEFAULT_SOURCE -D_FORTIFY_SOURCE=1 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/Users/emmanuelliu/connectedhomeip/examples/lighting-app/nrfconnect/main/include -I/Users/emmanuelliu/connectedhomeip/examples/lighting-app/lighting-common -I/Users/emmanuelliu/connectedhomeip/zzz_generated -I/Users/emmanuelliu/connectedhomeip/zzz_generated/app-common -I/Users/emmanuelliu/connectedhomeip/zzz_generated/lighting-app -I/Users/emmanuelliu/connectedhomeip/examples/platform/nrfconnect/util/include -I/Users/emmanuelliu/connectedhomeip/examples/platform/nrfconnect/app/include -I/Users/emmanuelliu/connectedhomeip/examples/lighting-app/nrfconnect/build/gen/app-codegen/cpp-app -I/Users/emmanuelliu/connectedhomeip/examples/lighting-app/nrfconnect/build/gen/app-zapgen/zapgen/app-templates -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrf/drivers/mpsl/clock_control -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr/include -I/Users/emmanuelliu/connectedhomeip/examples/lighting-app/nrfconnect/build/zephyr/include/generated -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr/soc/arm/nordic_nrf/nrf52 -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr/lib/libc/newlib/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr/lib/util/fnmatch/. -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr/soc/arm/nordic_nrf/common/. -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr/subsys/bluetooth -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr/subsys/net/lib/sockets/. -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr/subsys/settings/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrf/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrf/lib/multithreading_lock/. -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrf/subsys/bluetooth/controller/. -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr/drivers/flash -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrf/tests/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr/modules/openthread/platform/. -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/modules/hal/cmsis/CMSIS/Core/Include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/modules/hal/nordic/nrfx -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/modules/hal/nordic/nrfx/drivers/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/modules/hal/nordic/nrfx/mdk -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr/modules/hal_nordic/nrfx/. -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/modules/crypto/tinycrypt/lib/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/modules/lib/zcbor/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/nfc/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/mpsl/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/mpsl/include/protocol -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/openthread/include -I/Users/emmanuelliu/connectedhomeip/examples/lighting-app/nrfconnect/build/modules/nrfxlib/nrfxlib/nrf_security/src/include/generated -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/nrf_security/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/nrf_security/include/mbedtls -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/nrf_security/include/psa -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/mbedtls/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/mbedtls/include/mbedtls -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/mbedtls/include/psa -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/mbedtls/library -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/nrf_security/../crypto/nrf_oberon/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/nrf_security/../crypto/nrf_oberon/include/mbedtls -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/nrf_security/../crypto/nrf_oberon/include/psa -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/crypto/nrf_oberon/include/mbedtls -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/crypto/nrf_oberon/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/nrf_802154/driver/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/nrf_802154/sl/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/softdevice_controller/include -I/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr/include/zephyr/posix -I/Users/emmanuelliu/connectedhomeip/src -I/Users/emmanuelliu/connectedhomeip/src/include -I/Users/emmanuelliu/connectedhomeip/third_party/nlassert/repo/include -I/Users/emmanuelliu/connectedhomeip/third_party/nlio/repo/include -I/Users/emmanuelliu/connectedhomeip/examples/lighting-app/nrfconnect/build/modules/connectedhomeip/gen/include -I/Users/emmanuelliu/connectedhomeip/examples/providers -isystem /Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/crypto/nrf_cc310_platform/include -Werror -Wno-error=maybe-uninitialized "-DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h>" -fno-strict-aliasing -Os -fcheck-new -std=c++14 -Wno-register -fno-exceptions -fno-rtti -imacros /Users/emmanuelliu/connectedhomeip/examples/lighting-app/nrfconnect/build/zephyr/include/generated/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mfp16-format=ieee --sysroot=/Users/emmanuelliu/chip-nordic-tools/gnuarmemb/arm-none-eabi -imacros /Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/Users/emmanuelliu/connectedhomeip/examples/lighting-app/nrfconnect=CMAKE_SOURCE_DIR -fmacro-prefix-map=/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/Users/emmanuelliu/chip-nordic-tools/nrfconnect/sdk-nrf=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs -std=gnu++17 -D_SYS__PTHREADTYPES_H_ -MD -MT CMakeFiles/app.dir/Users/emmanuelliu/connectedhomeip/src/app/util/attribute-storage.cpp.obj -MF CMakeFiles/app.dir/Users/emmanuelliu/connectedhomeip/src/app/util/attribute-storage.cpp.obj.d -o CMakeFiles/app.dir/Users/emmanuelliu/connectedhomeip/src/app/util/attribute-storage.cpp.obj -c /Users/emmanuelliu/connectedhomeip/src/app/util/attribute-storage.cpp
In file included from /Users/emmanuelliu/connectedhomeip/src/app/util/config.h:31,
                 from /Users/emmanuelliu/connectedhomeip/src/app/util/af.h:23,
                 from /Users/emmanuelliu/connectedhomeip/src/app/util/common.h:21,
                 from /Users/emmanuelliu/connectedhomeip/src/app/util/attribute-storage.cpp:18:
[... cut ...]
8. source scripts/bootstrap.sh
9. source scripts/activate.sh
10. source ./042122_install_nordic_tools.sh
11. python3 scripts/setup/nrfconnect/update_ncs.py --update
12. export ZAP_INSTALL_PATH=/Users/emmanuelliu/tools/zap-mac/
13. cd examples/lighting-app/nrfconnect/
14. west build -b nrf52840dk_nrf52840 --pristine always

### Bug prevalence

Today

### GitHub hash of the SDK that was being used

fc41449659582354b44ed92905a6dc6ef9c9c227

### Platform

nrf, python

### Platform Version(s)

Darwin

bzbarsky-apple, 2023-02-06:
> I've updated to latest zap from https://github.com/project-chip/zap/releases.

That's the issue.  You need to be using ZAP `v2023.01.31-nightly`.  If you use `v2023.02.03-nightly` (which is current tip), you will get the above error, until https://github.com/project-chip/connectedhomeip/pull/24862 merges.

In general, you should look in `scripts/zap.json` to see what version of ZAP to use.

closed 2023-02-06 by bzbarsky-apple

esliu-apple, 2023-02-07:
Thank you! I've figured it out under your instructions.