project-chip/connectedhomeip #24897: iOS project is not compiling "No package 'openssl' found"

andrewstang, 2023-02-07:
xcode log：Package openssl was not found in the pkg-config search path.
Perhaps you should add the directory containing `openssl.pc'
to the PKG_CONFIG_PATH environment variable
No package 'openssl' found
Could not run pkg-config.

 I use "export PKG_CONFIG_PATH="/usr/local/opt/openssl/lib/pkgconfig" in terminal, it's no problem, pkg-config can search openssl. but how to do in Xcode?

bzbarsky-apple, 2023-02-07:
@andrewstang Put that `export` line in your `~/.bashrc` (the relevant Xcode bits are run in bash)?  Or manually run bootstrap from the command line in your terminal so that Xcode is not trying to do it.

That said, the symlink approach described at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md#installing-prerequisites-on-macos should also work (including on an M1 Mac, as long as the paths are correct; it looks like in your case things are in /usr/local/opt, not /usr/local/lib or /opt/homebrew/opt/.., which may depend on the exact package manager you used to install openssl).

bzbarsky-apple, 2023-02-07:
That said, the fact that this depends on openssl at all is broken.  I'll see if I can fix that.

andrewstang, 2023-02-08:
@bzbarsky-apple I pull down new repos, but it still fails to compile. build log 
[Build CHIP Tool App_2023-02-08T19-32-57.txt](https://github.com/project-chip/connectedhomeip/files/10685040/Build.CHIP.Tool.App_2023-02-08T19-32-57.txt)

bzbarsky-apple, 2023-02-08:
@andrewstang Is that with the `export PKG_CONFIG_PATH` bits added to your `.bashrc`?  If not, then "pull down new repos" is not going to change anything, since https://github.com/project-chip/connectedhomeip/pull/24901 is not merged yet.

closed 2023-02-08 by yufengwangca

bzbarsky-apple, 2023-02-08:
@andrewstang OK, _now_ #24901 is merged.  Try tip?

andrewstang, 2023-02-09:
@bzbarsky-apple build success！！！thanks a lot

bzbarsky-apple, 2023-02-09:
@andrewstang No problem, and thank you for filing this!