project-chip/connectedhomeip #24907: [Feature] Need a Way to Report EventList for Bridged Nodes

emargolis, 2023-02-07:
### Feature description

Refer to the following comment for more details:
https://github.com/project-chip/connectedhomeip/pull/24862#discussion_r1099155938

### Platform

core

bzbarsky-apple, 2023-07-06:
This is required to be able to bridge things properly.

MitterdorferMathias, 2023-11-02:
Hello,

We ported our feature complete bridge implementation to Matter 1.2. 
There we would need this feature to be able to fullfil the requirement:
`Bridged Device Basic Information Cluster -> Events -> Reach­ableChanged`.

What is the recommended way to proceed now?
- Downgrade to Matter 1.1
- Waiting for this ticket to be closed
- ...

bzbarsky-apple, 2023-11-02:
@MitterdorferMathias That requirement is not really part of the spec, because EventList is provisional.  It must not be included in actual devices to be certified for now, until it leaves provisional status.

MitterdorferMathias, 2023-11-03:
@bzbarsky-apple In Matter 1.2 the provisional flag was removed from `EventList`. See:
`23-27349-003_Matter-1.2-Core-Specification.pdf (Revision 3) > 7.13. Global Elements`

bzbarsky-apple, 2023-11-13:
@MitterdorferMathias That was actually an error, looks like: people thought it would leave provisional status, it did not, and they forgot to put the note back.  EventList is still provisional and not certifiable for the moment.

I filed https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/8502

xvortep, 2024-03-21:
Greetings, are there any updates on this? Will there be certification issues regarding reading eventList attributes of bridged clusters?
If so, can you provide some guidance on how to handle these attributes?
As I can see, they can be added to EmberAfCluster in this format

```
const chip::EventId brbEventList [] = {
 BridgedDeviceBasicInformation::Events::ReachableChanged::Id,
 chip::kInvalidEventId,
};

EmberAfCluster BridgedDeviceBasicInformation {
  .clusterId = BridgedDevicebasicInformation::Id,
  .attributes = ...,
  .attributeCount = ...,
  .clusterSize = ...,
  //....
  .eventList = brbEventList,
  .eventCount = ArraySize(brbEventList) - 1;
};

```

However, read requests to that attribute fail.

Thanks.

RDMChristoph, 2024-06-20:
So still for 1.2 Certification it is not required to support EventList for Bridged Device Basic Information Cluster?? In addition of course subscribing will return an UNSOPPORTED_EVENT is also not an issue for certification??

Is there an official statement confirming this?

robszewczyk, 2024-06-21:
I chair the Matter Technical Subgroup, I can confirm EventList was not required to be supported. We went through some gyrations around the implementation of the EventList.  It looked like we were about ready to land EventList in 1.2 and the spec reflected the attribute being marked non-provisional.  In the last steps of getting the 1.2 release out, I missed the fact that EventList implementation did not in fact get merged into the SDK in a broadly usable fashion; and should have remained provisional.  In 1.3 release, the provisional marking is back on EventList (see section 7.13. Global Elements in the Matter Core Specification).

closed 2024-06-21 by robszewczyk