project-chip/connectedhomeip #24913: [BUG] [ESP32-C3][lock-app] kernel panic

mmt-10, 2023-02-08:
### Reproduction steps

```bash
cd ${MATTER}/examples/lock-app/esp32
idf.py set-target esp32c3
idf.py build
idf.py erase_flash flash
```
esp32-c3 shows kernel panic and reboot.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 2e0737c66ec95342abc06762a85aa53b148ef83e (HEAD -> master, origin/master, origin/HEAD)

### Platform

esp32

### Anything else?

### Log
[20230208-esp32.txt](https://github.com/project-chip/connectedhomeip/files/10681678/20230208-esp32.txt)

### Environment
Host OS : Ubuntu 20.04
ESP-IDF : [ESP-IDF Release v4.4.4](https://github.com/espressif/esp-idf/releases/tag/v4.4.4)
ZAP :  [v2023.02.03-nightly](https://github.com/project-chip/zap/releases/tag/v2023.02.03-nightly)

bzbarsky-apple, 2023-02-08:
From the log: 
```
E (1448) chip[DL]: Chip stack locking error at './third_party/connectedhomeip/src/app/util/attribute-storage.cpp:510'. Code is unsafe/racy
```
That assert is at the top of `emAfReadOrWriteAttribute`.

@mmt-10 since you have a working and broken changeset, would you be willing to bisect to find out what changeset introduced the problem?

@shubhamdp @dhrishi

bzbarsky-apple, 2023-02-08:
Alternately, @mmt-10 if there is a way to get a stack to the assert, that would pretty much immediately pinpoint the problem, since the problem is some code that is trying to access attributes on the wrong thread.

mmt-10, 2023-02-09:
I assume there is a problem with the change after commit id bba4182d61d56510f628307aff9844b75fb27b0d at Jan/20.

I'm sorry, I do not know how do I debug it.

shubhamdp, 2023-02-09:
@mmt-10 I can reproduce this issue, will look into it.

dhrishi, 2023-04-04:
Fixed with #25895

closed 2023-04-04 by dhrishi