project-chip/connectedhomeip #24915: Unable to pair homekit with matter board.

fufenxiang, 2023-02-08:
**Problem**
I'm trying to add a matter board use ipad and homepodmini. BLE pairing always fail with following logs:

[00:01:54.714]Releasing end point's BLE connection back to application.
[00:01:54.720]HandleChipConnectionReceived failed, err = d
[00:01:54.725]failed handle new chip BLE connection, status = d

iOS version: 16.3
homepodmini version: 16.3
matter tag: v1.0.0

The complete log is shown below：
[homepodmini-pair-fail.txt](https://github.com/project-chip/connectedhomeip/files/10682951/homepodmini-pair-fail.txt)

Thank you for any suggestion.

tomasmcguinness, 2023-02-08:
What sort of matter board are you using?

Not sure how the HomePod is coming into play here as it looks like the HomeKit is trying to join the device to a WiFi network. 

The fail safe timer is expiring as it looks like the WiFi network wasn't joined.

Is your iPad connected to a WiFi network?

bzbarsky-apple, 2023-02-08:
What the log shows is that the device gets a ScanNetworks command:
````
[13:17:37.340]收←◆[00:00:54.135]>>> [E:3739r M:69558647] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
````
and then actually does a network scan:
```
[00:00:54.215]--wifi_ctrl_iface--vif[0]:flush_bss 0

[00:00:54.219]--scan 
...
[00:00:54.263][WiFi_I]45 scan start
...
[00:00:55.667][WiFi_I]scan finish
```

but at no point does the device send a _response_ to the ScanNetworks command.  It doesn't even send an ack, as far as I can tell (which is very odd, given the ordering of the code at https://github.com/project-chip/connectedhomeip/blob/v1.0-branch/src/app/clusters/network-commissioning/network-commissioning.cpp#L274-L275).

@fufenxiang Why is the "matter board" never responding to the ScanNetworks?

fufenxiang, 2023-02-09:
@tomasmcguinness The dev board is our company's self-developed devkit, it's func just like esp32, during process of pairing, both ipad and homepodmini connect wifi.

fufenxiang, 2023-02-09:
@bzbarsky-apple thanks for your reply. I test pairing process again and find network-commissioning.cpp#L274 is called. In addtion, i tried to pair with ipad ios16.1, the device was successfully added to homekit. I'm wondering if ios version lead  to this problem, what do i need to do to locate the cause.

bzbarsky-apple, 2023-02-09:
> and find network-commissioning.cpp#L274 is called.

Ok, but is the ack packet actually being sent?  Again, the log from the "matter board" shows it is _not sending the packet_.

> i tried to pair with ipad ios16.1, the device was successfully added to homekit.

Does that case do a ScanNetworks?

> what do i need to do to locate the cause.

Figure out why the "matter board" is not sending a response to the ScanNetworks command.

fufenxiang, 2023-02-13:
@bzbarsky-apple Thanks for your advice, this problem has been solved. The Scan Done Event is not triggered because of code error of network scan callback.

closed 2023-02-13 by fufenxiang