project-chip/connectedhomeip #24920: [BUG][TC-OO-2.3] OffWaitTime doesn't decrement after transitioning to Delayed Off state

marius-alex-tache, 2023-02-08:
### Reproduction steps

Regarding on/off operation state machine, transitioning to state `Delayed Off` doesn't start decrementing `OffWaitTime`. Instead, it sets its value to 0.

Steps to reproduce:
1. Commission DUT
2. ```./chip-tool onoff on 1 1```
3. ```./chip-tool onoff on-with-timed-off 1 300 300 1 1```
4. ```./chip-tool onoff off-with-effect 0 0 1 1```
5. ```./chip-tool onoff read off-wait-time 1 1```

According to the specification, while in the `Timed On` state:
-  any command which causes the `OnOff` attribute to be set to `FALSE` should trigger a transition to `Delayed Off`, which starts decrementing the `OffWaitTime` value.
- if the `OnTime` value reaches zero while in `Timed On`, then `OffWaitTime` is set to zero aswell. This logic is implemented inside `OnOffServer::updateOnOffTimeCommand` from `on-off-server.cpp`. However, it seems that somehow this code path is also reached in the first bullet. I've noticed this by commenting the line `OnOff::Attributes::OffWaitTime::Set(endpoint, 0);`. The DUT logs now show how `OffWaitTime` is decremented in the described scenario.

I'm not sure what's the issue here, but I think it has to do with the relationship between `on-off` and `level-control` clusters when on off lighting feature is enabled. Please let me know if you need additional details.

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

7e69c66bb44c91fe0b9bed7c7b8d272d3208c8dd

### Platform

k32w

bzbarsky-apple, 2023-02-08:
@jmartinez-silabs

marius-alex-tache, 2023-02-16:
Example: https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/nxp/k32w/k32w0
ZAP file: https://github.com/project-chip/connectedhomeip/blob/master/examples/lighting-app/nxp/zap/lighting-on-off.zap

Indeed, the ZAP file is different than the default one.

jmartinez-silabs, 2023-02-16:
Ok thanks i will try to compare both zap and maybe try with your config

PR #25172 (2023-02-19): [OnOff] Fix OffWaitTime delayed state off
Fix #24920

There is a timing window where `OnOff::Attributes::OnTime:` is set to `0` in `offWithEffectCommand` or in `setOnOffValue`  and the cluster timer call back -> `onOffWaitTimeOffEventHandler` -> `updateOnOffTimeCommand` is run before the `Attributes::OnOff::Set` off is done in `setOnOffValue`. This problematic timing window is exacerbated by `emberAfOnOffClusterLevelControlEffectCallback` delaying the Off state.

Fix entails reseting OnOff::Attributes::OnTime only in `setOnOffValue` and only after the OnOff state is finally set to off

Tested  [... cut]
closed 2023-02-19 by yufengwangca