project-chip/connectedhomeip #24928: [BUG] [TC-CADMIN-1.9] Commission window closed before 20th failed attempts

yinyihu-silabs, 2023-02-08:
### Reproduction steps

Steps to reproduce:
1. TH1 commissions DUT
2. Opens ECM commission windows with 900 second timeout window
3. TH2 commissions DUT with wrong manual pairing code and read commission WindowStatus.
4. Repeat 20 times, expect commission window to close.

From commission window status, the window is often closed within first 5-10 attempts, before the 20th attempts.
[tc-cadmin-1.09-allclusters.txt](https://github.com/project-chip/connectedhomeip/files/10689901/tc-cadmin-1.09-allclusters.txt)

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

5a845c658

### Platform

efr32, other

bzbarsky-apple, 2023-02-08:
@yinyihu-silabs What does the server log look like?

But looking at the log above, between this bit:
```
==================== wrong code #01
```
and this bit:
```
==================== wrong code #02
```
there are 8 PASE connection attempts to various IPs:
```
[1675864347.907450][3230:3232] CHIP:CTL: Attempting PASE connection to UDP:[fe80::64d8:4ff:fe5e:92f0%veth14c018a]:5540
[1675864348.285741][3230:3232] CHIP:CTL: Attempting PASE connection to UDP:[fe80::9049:e9ff:fef2:63c%vethca4efc5]:5540
[1675864349.015219][3230:3232] CHIP:CTL: Attempting PASE connection to UDP:[fe80::f47a:6dff:fe37:e00f%vethe260a32]:5540
[1675864351.020788][3230:3232] CHIP:CTL: Attempting PASE connection to UDP:[fe80::c836:b3ff:fec4:487d%veth2ed6360]:5540
[1675864355.024909][3230:3232] CHIP:CTL: Attempting PASE connection to UDP:[fe80::c0af:cff:fea5:b01a%veth31066fa]:5540
[1675864363.037357][3230:3232] CHIP:CTL: Attempting PASE connection to UDP:[fe80::1%br-a2133de8dca8]:5540
[1675864364.040166][3230:3232] CHIP:CTL: Attempting PASE connection to UDP:[fe80::42:33ff:fe4c:4e03%br-4d5672c76f59]:5540
[1675864366.043114][3230:3232] CHIP:CTL: Attempting PASE connection to UDP:[fe80::e65f:1ff:fe42:5508%wlan0]:5540
```
which all correspond to the same instance name and hostname, so the same device:
```
[1675864347.907039][3230:3232] CHIP:DIS: 	Hostname: E45F014255080000
[1675864347.907267][3230:3232] CHIP:DIS: 	Instance Name: AFE44B6C7B40BBF8
[1675864348.285340][3230:3232] CHIP:DIS: 	Hostname: E45F014255080000
[1675864348.285551][3230:3232] CHIP:DIS: 	Instance Name: AFE44B6C7B40BBF8
[1675864349.014516][3230:3232] CHIP:DIS: 	Hostname: E45F014255080000
[1675864349.014918][3230:3232] CHIP:DIS: 	Instance Name: AFE44B6C7B40BBF8
[1675864351.020369][3230:3232] CHIP:DIS: 	Hostname: E45F014255080000
[1675864351.020592][3230:3232] CHIP:DIS: 	Instance Name: AFE44B6C7B40BBF8
[1675864355.024543][3230:3232] CHIP:DIS: 	Hostname: E45F014255080000
[1675864355.024745][3230:3232] CHIP:DIS: 	Instance Name: AFE44B6C7B40BBF8
[1675864363.036985][3230:3232] CHIP:DIS: 	Hostname: E45F014255080000
[1675864363.037185][3230:3232] CHIP:DIS: 	Instance Name: AFE44B6C7B40BBF8
[1675864364.039772][3230:3232] CHIP:DIS: 	Hostname: E45F014255080000
[1675864364.039972][3230:3232] CHIP:DIS: 	Instance Name: AFE44B6C7B40BBF8
[1675864366.042759][3230:3232] CHIP:DIS: 	Hostname: E45F014255080000
[1675864366.042949][3230:3232] CHIP:DIS: 	Instance Name: AFE44B6C7B40BBF8
```

From the point of view of the device this is _8_ failed commissioning attempts.  So after this happens 3 times the device will close the commissioning window.

bzbarsky-apple, 2023-02-08:
In particular, this command:
```
./chip-tool pairing code 2 34970212338 --timeout 20 --commissioner-name beta
```
should probably be:
```
./chip-tool pairing code 2 34970212338 --timeout 20 --commissioner-name beta --discover-once
```
if you want to only contact the device once per command invocation.

yinyihu-silabs, 2023-02-09:
@bzbarsky-apple 
`--discover-once true` flag works.
will follow up with test plan team to update this.
thank you!

closed 2023-02-09 by yinyihu-silabs