project-chip/connectedhomeip #24934: [Feature]

eling22, 2023-02-09:
### Feature description

I was recently using a chip-tool and noticed a feature called "group control." 
I success to bind a group and use switch to control the group, but I can't find a way to use phone direct send group message to the group, like following reference.
https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool#using-the-client-to-send-group-multicast-matter-commands
I was wondering if this feature is available in Android as well. If it is, could you please direct me to where I can find it? If not, I would like to suggest that you consider adding this feature to the Android platform.

### Platform

android

### Platform Version(s)

v1.0.0

yunhanw-google, 2023-02-17:
yes, we currently don't have it, we need to add generic IM command in java/jni layer so that we can send out group command in android chip-tool, it would happen after I complete this https://github.com/project-chip/connectedhomeip/issues/24931

yunhanw-google, 2023-09-27:
@eling22 we just complete the group feature in android chip-tool, please have the test. thanks

yunhanw-google, 2023-09-27:
https://github.com/project-chip/connectedhomeip/pull/29244

PR #29244 (2023-09-27): [Android] Support group message
Fix #29100 
Fix #25448 
Fix #25449 
Fix #24934
1. Add GroupDeviceProxy for sending groupId in each command (write, invoke).
2. Add GroupSetting API 
   (refer https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool/commands/group)
3. For supporting ipv6 multicast, I add IPV6_MULTICAST_IMPLEMENTED feature and restore some codes related "if_nameindex" function. (related #25856 )
closed 2023-09-27 by yunhanw-google