project-chip/connectedhomeip #24947: .matter files should reflect codegen reality: mandatory global cluster attributes are always supported

bzbarsky-apple, 2023-02-09:
ZAP codegen ignores the "included" state for mandatory global cluster attributes, since that's not a valid configuration.  So no matter what the .zap file says they are supported by servers.

It looks like .matter files reflect the contents of the .zap file for these attributes (at least based on what I am seeing in https://github.com/project-chip/connectedhomeip/pull/24937/files), not the actual reality that these are always supported.

andy31415, 2023-02-09:
Do we have a list of what mandatory global attributes are always included so I can match this in the IDL logic?
I may need to do the same in xml parsing logic ...

Is there a code location where this decision is made or is it already set up in the XML definitions somehow?

bzbarsky-apple, 2023-02-09:
The specific list of attributes involved here is:
```
  0xfff8, // GeneratedCommandList
  0xfff9, // AcceptedCommandList
  0xfffa, // EventList
  0xfffb, // AttributeList
```

On the server side, these are excluded from the metadata (because every cluster must have them) and the IM code special-cases them.  See https://github.com/project-chip/zap/blob/fd7e0bfbc72c3821053352730f02c75de229d969/src-electron/generator/helper-endpointconfig.js#L965 and its callsites in that file, and https://github.com/project-chip/connectedhomeip/blob/16a64b357293a9ef02c541ff743e3fabf057bc71/src/app/GlobalAttributes.h#L31 and its consumers.

PR #24951 (2023-02-10): Force global atrributes to be declared for every cluster in MatterIDL
Spect section `7.13 Global Elements` describes a set of elements that are mandatory for every cluster to have.

Fixes #24947 for the matter IDL bits (although java may still depend on zap, where handling seems to need specialization)

It also turns out that both JAVA and PYTHON rely on zap templating to have global attributes enabled. As a result I updated controller_clusters.zap to enable mandatory attributes (a lot of event list adding, some adding/enabling for a few other clusters). This resulted in java and python genera [... cut]
closed 2023-02-10 by andy31415