project-chip/connectedhomeip #24957: Hi, for Android CHIP Tool  demo, how to get all ".so" lib files?

Steven1937, 2023-02-10:
### Documentation issues

Hi, for Android CHIP Tool  demo, how to get all ".so" lib files?

jonsmirl, 2023-02-11:
https://github.com/google-home/sample-app-for-matter-android/issues/76

how these libs were built?

We simply build the examples for the various android targets,

./scripts/build/build_examples.py \
  --target android-arm-chip-tool \
  --target android-arm64-chip-tool \
  --target android-x86-chip-tool \
  --target android-x64-chip-tool \
  build
And then copy the libraries found in src/android/CHIPTool/app/libs

Steven1937, 2023-02-13:
Hi, When building the whole project, or when building the instructions you give,  always fails due to missing files,  seems cannot clone all the files and full environment,  could u please send me the .so files directly?  my email is 1454841908@qq.com, thank you very much

Steven1937, 2023-02-13:
> 
Hi, When building the whole project, or when building the instructions you give, always fails due to missing files, seems cannot clone all the files and full environment, could u please send me the .so files directly? my email is [1454841908@qq.com](mailto:1454841908@qq.com), thank you very much

Steven1937, 2023-02-13:
> [google-home/sample-app-for-matter-android#76](https://github.com/google-home/sample-app-for-matter-android/issues/76)
> 
> how these libs were built?
> 
> We simply build the examples for the various android targets,
> 
> ./scripts/build/build_examples.py --target android-arm-chip-tool --target android-arm64-chip-tool --target android-x86-chip-tool --target android-x64-chip-tool build And then copy the libraries found in src/android/CHIPTool/app/libs

Hi, When building the whole project, or when building the instructions you give, always fails due to missing files, seems cannot clone all the files and full environment, could u please send me the .so files directly? my email is [1454841908@qq.com](mailto:1454841908@qq.com), thank you very much

jonsmirl, 2023-02-13:
They are in the Google sample app project
https://github.com/google-home/sample-app-for-matter-android/tree/main/app/third_party/connectedhomeip/libs

Steven1937, 2023-02-13:
> They are in the Google sample app project https://github.com/google-home/sample-app-for-matter-android/tree/main/app/third_party/connectedhomeip/libs

Thank you very much

closed 2023-10-15 by stale[bot]