project-chip/connectedhomeip #24961: Synchronize two clients to control one device

stefano055415, 2023-02-10:
i have two clients (Phone App) using the same account.

with the first I carry out commissioning with a device, I would like to synchronize the second via cloud so that I can use the device without having to repeat the commissioning

Could you tell me the correct flow?

it is sufficient to share the NOC with the second client

bzbarsky-apple, 2023-02-10:
You need to make sure you transfer enough information to stand up the second app's node on the relevant fabric.  So
1.  Either the root certificate and a pre-generated NOC (and ICA cert if any are in use) or enough information to generate the operational certificate chain (root private key, root public key, root certificate matter-rcac-id, fabric id), depending on whether the second app will be doing commissioning too.
2. The IPK.

closed 2023-02-10 by bzbarsky-apple

DanijelBojcic, 2023-03-07:
We want to achieve the same, totally skipping the commissioning on the second phone.

@stefano055415  Did you solve this?

@bzbarsky-apple  Assume we have everything ( root certificate, NOC, IPK , IP, discriminator, passcode , ... anything) whats the fastest flow to pair the device?

bzbarsky-apple, 2023-03-07:
@DanijelBojcic I don't follow the question.  If you have all the relevant info, once you "pair" (that is commission) the device onto the fabric once, the second client on the same fabric does not need to do any "pairing".  It can just do operational lookup and talk to the device.

DanijelBojcic, 2023-03-07:
My concern is how to load all the information onto the second client. I can't seem to find any existing functions that can do this. Should I simply copy the KeyValueStore from the first client?

bzbarsky-apple, 2023-03-07:
> My concern is how to load all the information onto the second client.

That's up to you; it depends on how you are storing that information, etc.  For example, you probably want to copy the root cert and the IPK, but for the operational cert you presumably want to issue the second client a cert for the same fabric but a different node id.

There are no "existing functions" to do this because it depends on your exact PKI setup.

8TRobi8, 2023-03-20:
@bzbarsky-apple we would like to use Matter as a communication interface between our controller and the mobile apps which connect to it. We would like to avoid the commissioning process as much as we can. So the algorithm would be that for example the first mobile app who connects to the network commissions the device and shares the required information to the other applications, so store them in the cloud.

We tried to share the required information and also the NOC as well and use the same NOC on all mobile devices to connect to the controller. I don't know if it's a valid way or not, therefore I tried to validate it with chip-tool. I've run two chip-tools in parallel in two different windows. Let's call the first chip-tool as 'A', the second one as 'B'.

1. chip-tool 'A' subscribed to an attribute on the controller:
`onoff subscribe on-off 1 100 1 4`
2. chip-tool 'B' subscribed with the same NodeId to the same attribute on the controller:
`onoff subscribe on-off 1 100 1 4 --keepSubscriptions true`

At this point, chip-tool 'A' hasn't got any error/event/information from the server, but its subscription doesn't work. BUT, if chip-tool 'A' re-subscribes within the timeout window (maxInterval parameter from the subscription above which was 100s), then the subscription works on both chip-tools. `onoff subscribe on-off 1 100 1 4 --keepSubscriptions true`

It means, two chip-tools - so two clients - are connected to the server with the same NOC, but with different Network addresses (in my case the IP is the same, but different ports are used).

My question would be: Is this the intended behavior? So, should it work with two clients with the same NOC? (then the issue is that the first client has to re-subscribe) Or, it shouldn't work at all that two clients are connected with the same NOC and this is not permitted?

bzbarsky-apple, 2023-03-21:
> the first mobile app who connects to the network commissions the device and shares the required information to the other applications, so store them in the cloud

This is all fine and doable, yes.

> and use the same NOC on all mobile devices to connect to the controller

This is not a great idea for the following reason: if you are using the same NOC on all of them, you are using the same node ID for all of them.  The other side of the interaction can get very confused when it needs to "send a message to this node id" (which is what the spec requires) but the node id conceptually corresponds to multiple IPs/ports.

>  then the subscription works on both chip-tools

This basically happens to work by accident, sometimes, and is very much not guaranteed.   And anything that causes either of the clients to establish a new CASE session will make the subscription for the other client stop working.  You observred that when B established its subscription initially and A's subscription stopped working.

> It shouldn't work at all that two clients are connected with the same NOC and this is not permitted?

It's not expected to work, but might work by accident, in some circumstances.  The set of circumstances where it happens to work is not clearly defined and not guaranteed to stay the same as changes happen to the underlying core code.

Per spec, a node id is supposed to map to exactly one thing on the network.  That thing's ip+port can change (most simply: dhcp assigns a new IP), but you really should not have multiple things all using the same node id at the same time; that will not work well.