project-chip/connectedhomeip #24964: [BUG] Invalid TLV tag

mustafayuce33, 2023-02-10:
### Reproduction steps

I test dynamic bridge app. I can successfully added on\off device to Apple Home. But if I set turn on/off the device then I get this error;

CHIP:DL: ../../third_party/connectedhomeip/src/lib/core/TLVWriter.cpp:637: CHIP Error 0x00000024: Invalid TLV tag

Please see the full error logs below; 

```
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:EM: >>> [E:59636r M:145551799] (S) Msg RX from 1:136DCAECB280E746 [AD3E] --- Type 0001:08 (IM:InvokeCommandRequest)
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:EM: Handling via exchange: 59636r, Delegate: 0x7e51ec
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG: InvokeRequestMessage =
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG: {
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:         suppressResponse = false,
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:         timedRequest = false,
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:         InvokeRequests =
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:         [
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:                 CommandDataIB =
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:                 {
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:                         CommandPathIB =
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:                         {
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:                                 EndpointId = 0x4,
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:                                 ClusterId = 0x6,
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:                                 CommandId = 0x1,
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:                         },
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:                         CommandFields =
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:                         {
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:                         },
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:                 },
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:         ],
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG:         InteractionModelRevision = 1
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG: },
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG: AccessControl: checking f=1 a=c s=0x136DCAECB280E746 t= c=0x0000_0006 e=4 p=o
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG: AccessControl: allowed
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG: Received command for Endpoint=4 Cluster=0x0000_0006 Command=0x0000_0001
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:ZCL: On/Off set value: 4 1
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DL: emberAfExternalAttributeReadCallback: endpoint=4 clusterId=6 attrId=0
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DL: ../../third_party/connectedhomeip/src/lib/core/TLVWriter.cpp:637: CHIP Error 0x00000024: Invalid TLV tag
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DL: Attribute access(interface) failed to read attribute 0, for endpoint 4 cluster 6
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:ZCL: ERR: reading on/off 1
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG: Command handler moving to [ Preparing]
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG: Command handler moving to [AddingComm]
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG: Command handler moving to [AddedComma]
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG: Decreasing reference count for CommandHandler, remaining 0
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:EM: <<< [E:59636r M:145430701 (Ack:145551799)] (S) Msg TX to 1:136DCAECB280E746 [AD3E] --- Type 0001:09 (IM:InvokeCommandResponse)
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:IN: (S) Sending msg 145430701 on secure session with LSID: 63877
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG: Command handler moving to [CommandSen]
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:DMG: Command handler moving to [AwaitingDe]
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:EM: >>> [E:59636r M:145551800 (Ack:145430701)] (S) Msg RX from 1:136DCAECB280E746 [AD3E] --- Type 0000:10 (SecureChannel:StandaloneAck)
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:EM: Found matching exchange: 59636r, Delegate: (nil)
Feb 10 10:26:25 project-HEHGGN project-matter[7243]: CHIP:EM: Rxd Ack; Removing MessageCounter:145430701 from Retrans Table on exchange 59636r
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/tree/2ce5f2d6e88a69e24ca8542cd03c233292755423

### Platform

raspi

bzbarsky-apple, 2023-02-10:
> CHIP:DL: emberAfExternalAttributeReadCallback: endpoint=4 clusterId=6 attrId=0

Where is this log coming from?  I don't see it in the source tree...

That said, what I recommend doing is breakpointing on that line where the error happens and seeing who it is that's trying to encode something with a context tag outside a list or struct.

@andy31415 can you please assign to @dnicoara ?

mustafayuce33, 2023-02-10:
That was my mistake sorry.

closed 2023-02-10 by mustafayuce33