project-chip/connectedhomeip #24965: [BUG][esp32] TC-RR-1.1 Operational Discovery fails

PSONALl, 2023-02-10:
### Reproduction steps

- Tested on TE1.1 commit 
- TH version- 2.7.1 
- Commissioning is complete but the test RR-1.1 fails after running python script TC_RR_1_1.py
- logs
[esp32.txt](https://github.com/project-chip/connectedhomeip/files/10706577/esp32.txt)
[th-rr-1.1.txt](https://github.com/project-chip/connectedhomeip/files/10706578/th-rr-1.1.txt)

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

7e69c66bb44c91fe0b9bed7c7b8d272d3208c8dd

### Platform

esp32

bzbarsky-apple, 2023-02-10:
Operational discovery failing is a side-effect.  The real issue seems to be this (in the th-rr-1.1-txt log):
```
[1676025914.623575][39:42] CHIP:SC: Failed to verify peer's MAC. This can happen when setup code is incorrect.
...
[1676025914.624945][39:42] CHIP:SC: Failed during PASE session setup: ../../src/crypto/CHIPCryptoPALOpenSSL.cpp:1440: CHIP Error 0x000000AC: Internal error
Failed to establish secure session to device: ../../src/crypto/CHIPCryptoPALOpenSSL.cpp:1440: CHIP Error 0x000000AC: Internal error
Failed to commission: ../../src/crypto/CHIPCryptoPALOpenSSL.cpp:1440: CHIP Error 0x000000AC: Internal error
[MatterTest] 02-10 10:45:14.628 INFO [Test] test_run_commissioning PASS
[MatterTest] 02-10 10:45:14.633 INFO Summary for test class CommissionDeviceTest: Error 0, Executed 1, Failed 0, Passed 1, Requested 1, Skipped 0
```

so the node is _NOT_ commissioned (probably because the setup code is in fact incorrect?) but the commissioning test somehow still passes (? @tcarmelveilleux ) and after that of course operational discovery will fail.

tcarmelveilleux, 2023-02-10:
```
[0;32mI (49258) chip[EM]: >>> [E:60634r M:116495337 (Ack:43897873)] (U) Msg RX from 0:522ACBC1C87FA96C [0000] --- Type 0000:22 (SecureChannel:PASE_Pake1)[0m
[0;32mI (49268) chip[EM]: <<< [E:60634r M:43897874 (Ack:116495337)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)[0m
[0;32mI (49288) chip[IN]: (U) Sending msg 43897874 to IP address 'UDP:192.168.20.170%st2:40785'[0m
[0;32mI (50108) chip[EM]: <<< [E:60634r M:43897875 (Ack:116495337)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:23 (SecureChannel:PASE_Pake2)[0m
[0;32mI (50118) chip[IN]: (U) Sending msg 43897875 to IP address 'UDP:192.168.20.170%st2:40785'[0m
[0;31mE (50128) chip[DL]: Long dispatch time: 868 ms, for event type 3[0m
[0;32mI (50218) chip[EM]: >>> [E:60634r M:116495338 (Ack:43897875)] (U) Msg RX from 0:522ACBC1C87FA96C [0000] --- Type 0000:40 (SecureChannel:StatusReport)[0m
[0;31mE (50228) chip[SC]: Received error (protocol code 2) during PASE process: 38[0m
[0;31mE (50228) chip[SC]: Failed during PASE session setup: 38[0m
[0;31mE (50238) chip[SVR]: Commissioning failed (attempt 18): 38[0m
[0;32mI (50238) chip[DIS]: Updating services using commissioning mode 1[0m
[0;32mI (50258) chip[DIS]: CHIP minimal mDNS started advertising.[0m
[0;32mI (50278) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1[0m
[0;32mI (50278) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 46FC26AF4F45AA1C.[0m
```

So a couple things:

- Error 0x38 is `CHIP_ERROR_INVALID_PASE_PARAMETER`
- This can happen on bad passcode
- The `th-rr-1.1.txt` log is *missing the arguments*. It is showing `$PASSCODE` and so we cannot tell what was being executed
- Yes it's a bug that `test_run_commissioning` passed here...

To assist in debugging:
- Please provide the full command line for the TC_RR_1_1.py, with all environment variables expanded
- Please tell us the passcode for the device (ESP32)

PSONALl, 2023-02-13:
1. Full command line: 
2. > rm -f admin_storage.json && python3 python_testing/TC_RR_1_1.py --discriminator $DISCRIMINATOR --passcode $SETUP_PASSCODE --commissioning-method on-network --paa-trust-store-path /paa_roots --storage-path admin_storage.json
3. $DISCRIMINATOR = 3840 $SETUP_PASSCODE = 20202021 
4.  passcode for ESP32 was 20202021

cletnick, 2023-02-17:
This was fixed in PR #24919 last week.

closed 2023-07-06 by PSONALl