project-chip/connectedhomeip #24966: [BUG][ESP32]TC[ACE-1.4] Details=PIXIT.ACE.APPENDPOINT is not found

PSONALl, 2023-02-10:
### Reproduction steps

- Tested on TE1.1 commit
- The script for running the test case fails with error: Details=PIXIT.ACE.APPENDPOINT must be included on the command line in the --int-arg flag as PIXIT.ACE.APPENDPOINT:<endpoint>, Extras=None
- Logs 
      - DUT [esp32.txt](https://github.com/project-chip/connectedhomeip/files/10706864/esp32.txt)
      - Test Harness [th.txt](https://github.com/project-chip/connectedhomeip/files/10706865/th.txt)

### Bug prevalence

EveryTime

### GitHub hash of the SDK that was being used

7e69c66bb44c91fe0b9bed7c7b8d272d3208c8dd

### Platform

esp32

bzbarsky-apple, 2023-02-10:
@tcarmelveilleux @cjandhyala

wqx6, 2023-02-13:
Please use the following command:
```
python3 python_testing/TC_ACE_1_4.py --int-arg PIXIT.ACE.APPENDPOINT:1 PIXIT.ACE.APPDEVTYPEID:{device-type-id-of-app-endpoint} --string-arg PIXIT.ACE.APPCLUSTER:OnOff  PIXIT.ACE.APPATTRIBUTE:StartUpOnOff --discriminator 3840 --passcode 20202021 --commissioning-method ble-wifi --paa-trust-store-path /paa_roots --storage-path admin_storage.json --wifi-ssid {ssid} --wifi-passphrase {password} --ble-interface-id 0
```

cjandhyala, 2023-06-26:
@PSONALl Could you pls file issue in scripts repo if it is still an issue with the test script

cjandhyala, 2023-06-26:
@PSONALl closing the issue for now. pls re-open if you think the issue still exists.

closed 2023-06-26 by cjandhyala