project-chip/connectedhomeip #24970: [BUG]TC-CGEN-2.4 PASESession timed out at step 3

PSONALl, 2023-02-10:
### Reproduction steps

- Tested on TE1.1 commit
- At step 3 - error recieved : PASESession timed out while waiting for a response from the peer
- Logs 
       - DUT: [esp32.txt](https://github.com/project-chip/connectedhomeip/files/10707670/esp32.txt)
       - TH: [th.txt](https://github.com/project-chip/connectedhomeip/files/10707671/th.txt)

### Bug prevalence

EveryTime

### GitHub hash of the SDK that was being used

7e69c66bb44c91fe0b9bed7c7b8d272d3208c8dd

### Platform

esp32

adamb-q, 2023-02-13:
I'm also seeing this on qpg platform

marius-alex-tache, 2023-03-06:
Also on k32w0.

PSONALl, 2023-07-06:
I am still facing this issue on updated TH version

PSONALl, 2023-07-11:
This is not an sdk issue, This is an issue with test harness, as the test passes with chip-tool
I have raised an issue on TH repo here: https://github.com/CHIP-Specifications/chip-certification-tool/issues/883
So closing this issue

closed 2023-07-11 by PSONALl