project-chip/connectedhomeip #24971: [BUG] [TC-CADMIN-1.9] Commissioning retry counter is incrementing faster than expected

adamb-q, 2023-02-10:
### Reproduction steps

Platform: qpg, chip-tool x64

Running CADMIN 1.9 shows that using ECM commissioning counter is incrementing faster than expected, which causes test failure earlier than on 20th retry. When using BCM the issue is not happening - 1 failure per 1 attempt. Attaching logs from CADMIN 1.9 and 1.10 for reference.

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

7e69c66bb44c91fe0b9bed7c7b8d272d3208c8dd

### Platform

other

### Platform Version(s)

qpg

### Anything else?

[TC_CADMIN_1_9.1675432562.log](https://github.com/project-chip/connectedhomeip/files/10707609/TC_CADMIN_1_9.1675432562.log)
[TC_CADMIN_1_9.1675432562_DUT.log](https://github.com/project-chip/connectedhomeip/files/10707610/TC_CADMIN_1_9.1675432562_DUT.log)
[[TC-CADMIN-1.10]_[061nwOu0]_CHIPTool_log.log](https://github.com/project-chip/connectedhomeip/files/10707614/TC-CADMIN-1.10._.061nwOu0._CHIPTool_log.log)
[TC_CADMIN_1_10.1675433327_DUT.log](https://github.com/project-chip/connectedhomeip/files/10707615/TC_CADMIN_1_10.1675433327_DUT.log)

bzbarsky-apple, 2023-02-10:
@adamb-q Is this a duplicate of https://github.com/project-chip/connectedhomeip/issues/24928 ?

adamb-q, 2023-02-13:
@bzbarsky-apple I applied the `--discover-once true` flag, and it looks better, but still the log output from chip-tool gives the correct output only on 1st attempt, I mean `Failed to verify peer's MAC. This can happen when setup code is incorrect.` The next steps differ, can I assume that it is correct behavior also? Attaching logs with the discover-once flag set.

[TC_CADMIN_1_9.1676278577_DUT.log](https://github.com/project-chip/connectedhomeip/files/10720752/TC_CADMIN_1_9.1676278577_DUT.log)
[TC_CADMIN_1_9.1676278577.log](https://github.com/project-chip/connectedhomeip/files/10720753/TC_CADMIN_1_9.1676278577.log)

bzbarsky-apple, 2023-02-13:
That sounds like https://github.com/project-chip/connectedhomeip/issues/22547 ?

adamb-q, 2023-02-14:
Yes, looks relevant. When waiting ~50s between retries test would pass, but it exceeds 900s and commissioning window closes.

PR #25176 (2023-02-21): Rewrite discover-once to really stop right after the first PASE sessi…
…on failure

#### Problem

The `discover-once` argument for `chip-tool` is not really stopping the PASE establishment. This PR update the code such that it really only tries a single ip .

I suspect that is should fix the issues related to `discover-once`, notably:
#fix #24971
#fix #22547
#fix #22407
closed 2023-02-21 by vivien-apple