project-chip/connectedhomeip #24981: [BUG] TE-1.1 TC-CADMIN-1.19 and CADMIN-1.20 commissioning fails with no memory error when repeated 6 times

PSONALl, 2023-02-10:
### Reproduction steps

- Tested on TE1.1 commit
- Logs :
    DUT [esp32.txt](https://github.com/project-chip/connectedhomeip/files/10710188/esp32.txt)
    TH [th.txt](https://github.com/project-chip/connectedhomeip/files/10710178/th.txt)

### Bug prevalence

EveryTime

### GitHub hash of the SDK that was being used

7e69c66bb44c91fe0b9bed7c7b8d272d3208c8dd

### Platform

esp32

bzbarsky-apple, 2023-02-11:
@PSONALl The server is required by spec to support 5 fabrics.  While it's allowed to support more, it does not have to, so having it respond with an error when trying to add a 6th fabric is quite reasonable.  What behavior were you expecting, exactly, and why?

PSONALl, 2023-02-11:
TC-CADMIN-1.20 and TC-CADMIN-1.19 expect that the DUT should be commissioned a "number of supported fabrics"(which is 16 as per TE-1.1 verification steps document) 
If the number of supported fabrics = 5 then this might be the issue with the test plan document

bzbarsky-apple, 2023-02-11:
> which is 16 as per TE-1.1 verification steps document

The number of supported fabrics should be read from the device, which exposes that as an attribute.  Please file a test plan verification step issue to get that fixed.

closed 2023-02-11 by bzbarsky-apple